llvm/circt #6706: [HW] Module port accessors and modifiers bugs

youngar, 2024-02-15:
A long time ago, HW module op would store its inputs and outputs using a function type.  This lacked the ability to intermix input and output ports the way verilog allows, and so we overhauled the system.  We added a `ModuleType` which holds an array of all ports, storing the name, type, and direction of each port.  It seems that different parts of our code have not been updated properly with the new system.

---
In the HWModuleOpInterface, the `setInputNames` makes an assumption that the port names of inputs are stored before all outputs.  When you update the input port names, it will replace replace all the existing input port names, but then effectively partially sort by input direction before commiting the names.  This is happening because the code assumes that all output ports come after the input ports, which is simply not true.

https://github.com/llvm/circt/blob/a3e993cb387a08e686b0df1d7f5b599bca0b3751/include/circt/Dialect/HW/HWOpInterfaces.td#L135-L140

This issue can be found in `setInputNames`, `setOutputNames`, `setAllInputAttrs`, `setAllOutputAttrs`, `setInputLocs`, `setOutputLocs`.

---

`modifyModuleArgs` is used to insert and delete ports from modules.  Due to historical reasons, it treats input and output ports differently.  I believe that this function will successfully modify the input and output ports as requested, but then follow up by sorting them by direction.  I don't think this is on purpose, and would argue it would make sense to retain the original port order. I.e. the input and output indices should be changed in to a global port index.  An ever better API here would be to change this function so that it used a single list of port modifications for both inputs and outputs.

https://github.com/llvm/circt/blob/a3e993cb387a08e686b0df1d7f5b599bca0b3751/lib/Dialect/HW/HWOps.cpp#L582

---

https://github.com/llvm/circt/blob/a3e993cb387a08e686b0df1d7f5b599bca0b3751/include/circt/Dialect/HW/HWStructure.td#L458-L462

HWInstanceOp `getPortIdForInputId` and `getPortIdForOutputId` are wrong, and should just forward the question to the ModuleType

https://github.com/llvm/circt/blob/a3e993cb387a08e686b0df1d7f5b599bca0b3751/lib/Dialect/HW/HWTypes.cpp#L825

---

https://github.com/llvm/circt/blob/a3e993cb387a08e686b0df1d7f5b599bca0b3751/include/circt/Dialect/HW/HWOpInterfaces.td#L178C28-L200

This is not a bug, but this code could be faster.  `getPortIdForInputId` and `getPortIdForOutputId` involve a serial walk through all ports.  It would be faster to just walk all ports, filtering out the unneeded directions. Same issue in `getInputLocs, `getInputLocsAttr`, `getOutputLocs`, `getOutputLocsAttr`,

darthscsi, 2024-02-15:
| HWInstanceOp getPortIdForInputId and getPortIdForOutputId are wrong, and should just forward the question to the ModuleType

This breaks hw.module parallelization.

youngar, 2024-02-15:
I forgot that InstanceOp doesn't hold a ModuleType.  I guess there is no way to implement these functions as is, and so they should just be deleted.

darthscsi, 2024-02-15:
The intention is to get rid of getInput* and getOutput*.  Nothing in HW is suppose to use them, everything there should be using port based versions.  Thus these aren't suppose to impact performance for migrated dialects.

darthscsi, 2024-02-16:
Also referenced (commit message wrong) in 9e0c1696f3caef4059c65774ad6b8efee91d9d9e