llvm/circt #6715: [FIRRTL] Remove "ref foo is <string>" syntax

seldridge, 2024-02-18:
Now that probes have been operational for some time, we no longer need to accept non-standard formats that allow string paths to be exposed from inside external modules.  This has been fully subsumed with the FIRRTL ABI that produces a ref file of Verilog macros.

E.g., there are tests that accept the following in `parse-basic.fir`:

```
  extmodule RefExtMore :
    input in : UInt<1>
    output r : Probe<UInt<1>>
    output data : UInt<3>
    output r2 : Probe<{a : UInt<3>}[3]>
    ref r2 is "in"
    ref r is "path.to.internal.signal"
```

These tests, parsing, and any handling of "internalpaths" can be removed.

dtzSiFive, 2024-02-18:
These are and have been in the FIRRTL spec since probes/ref types were
introduced.

You could think of them as ref literals, and pre-date the ABI.

They may have utility for use with non-FIRRTL-defined modules, e.g.
https://github.com/chipsalliance/chisel/pull/3486 .

Having Chisel emit "ABI" files directly itself for this use case is a
pain.  And arguably the wrong level but makes it not part of the
declaration which has strengths.

Perhaps a friction that some tooling could assist or just a burden to take
on anyway (as that linked PR did for an earlier ABI, sorry if that broke
with the ABI change by the way!)?

Anyway they're not currently used but they're not non-standard so uh please
do not remove them until they are.

They're also the only way to XMR into/through an intrinsic, which is ...
fun 🙃..  but was requested when I touched this last, so a consideration
for this direction.

dtzSiFive, 2024-02-20:
> need to accept non-standard formats

Fix?

---- 

As for the rest, wrote from phone so perhaps some tone was lost -- "here are other things you might also not be aware of".

Having a stable versioned ABI for these things sounds good, but still presently it's a bit painful to roll code to emit the right macros for.  That was not mean to say ref statements are better or even good.  And "wrong layer" I mean that it encodes what's known at a high-level directly into low-level implementation details, making it impossible to understand this information without understanding the verilog macros.  "Wrong" is maybe hasty but hopefully you see what I mean.

This means FIRRTL itself cannot be used to represent a circuit's probes into non-FIRRTL bits, without some sort of Verilog preprocessor + file scan or whatever, and this may be undesirable if wanting the extmodule to be a simulated thing or whatever.

Practically not a big loss but there's a more complete explanation.  This is in some small part also suggesting maybe having cross-module information encoded as Verilog macros has some downsides, said another way?

Your reasoning about wanting the ABI used makes sense, given that ship has sailed.  Thanks.

Anyway as mentioned, the pain can be helped by maintaining Chisel code to understand and emit the FIRRTL module ABI -- or other tooling that helps do this when interacting with non-Chisel-generated modules.

---

Dropping sounds good :+1: .

One fewer thing to worry about and maintain / forget about and have interact poorly when someone tries to use it.