llvm/circt #6748: [FIRRTL] Correct bindfile output for layers

rwy7, 2024-02-26:
As mentioned [here](https://github.com/llvm/circt/pull/6733#discussion_r1498385896), when firtool lowers layers to bound-in instances, the bind statements are written to the wrong file.

The bind convention, as specified in [the ABI](https://github.com/chipsalliance/firrtl-spec/blob/main/abi.md#bind-lowering-convention), indicates that each public module should have its own bindfile for each layer:
```
filename = "layers_" , module , "_", root , { "_" , nested } , ".sv" ;
```

The actual output of firtool, however, emits one file per layer, with the format:
```
filename = "layers_" , circuit , "_", root , { "_" , nested } , ".sv" ;
```

dtzSiFive, 2024-02-26:
cc #6716 .

rwy7, 2024-02-26:
duplicate

closed 2024-02-26 by rwy7