llvm/circt #6752: [FIRRTL] Generate filelist for each public module

tymcauley, 2024-02-26:
According to the FIRRTL ABI spec:

> Each public module in a circuit shall produce a filelist that contains the filename of the file containing the public module and any necessary files that define all public or private module files instantiated under it.

Source here: https://github.com/chipsalliance/firrtl-spec/blob/fcd97a59319eab2553afc301d13ccc9af965a2b8/abi.md#L48

However, if you have this FIRRTL:

```firrtl
FIRRTL version 4.0.0
circuit Foo :
  public module Bar :

  public module Foo :
    inst bar of Bar
```

And run this command:

```
$ firtool --split-verilog Foo.fir -o out
```

Then in the `out` directory, you'll get `Foo.sv`, `Bar.sv`, and `filelist.f`. The filelist is for module `Foo`, and it contains:

```
$ cat out/filelist.f
Bar.sv
Foo.sv
```

According to the ABI spec, we should get `out/filelist_Foo.f` and `out/filelist_Bar.f`, right?

Possibly related to https://github.com/llvm/circt/issues/6716.

seldridge, 2024-03-03:
You're correct in that this is not aligning with the spec. We keep discussing an alternative to this, and changing the spec to align with it, of a "manifest" which is some file that includes enough information to generate these filelists. It would be something that represents the collection of instance graphs in the design, enumerates which modules are public, and we would then ship some lightweight tools to generate filelists from it.

One of the complications here is that a filelist (to most tools consuming it) is a list of command line arguments. By introducing this manfiest and tools that process it to produce command line arguments, we'd make this more generic for a tool that can't handle this basic filelist format.

@dtzSiFive and @rwy7 have thought more about this.

seldridge, 2024-03-04:
See https://github.com/chipsalliance/firrtl-spec/issues/189 for ideas around FIRRTL manifests.

tymcauley, 2024-08-22:
Just checking in, I see that https://github.com/chipsalliance/firrtl-spec/issues/189 was resolved, with the conclusion that manifests aren't the path forward. Would the idea be to place all outputs in a single directory (as is the case now), but then enable a post-processing step to create a directory-and-filelist for each public module? Would that post-processing step be part of `firtool`?