llvm/circt #6793: [Ibis] Uses of replaceAllUsesWith in DialectConversion can no longer replace uses in replaced/erased ops.

mikeurbach, 2024-03-06:
In the latest LLVM bump (https://github.com/llvm/circt/pull/6784), we are including some new, tighter assertions in DialectConversion. We want to get this bump in, since it includes a use-after-free bugfix in upstream MLIR (this currently manifests in integration test failures).

I took care of updating a couple passes that ran afoul of these tighter assertions, but there are some uses of `rewriter.replaceAllUsesWith` in Ibis transforms that now assert.

To summarize, after https://github.com/llvm/llvm-project/commit/6008cd40b7bbdc66555550c2e38648d5ce99cc78, hooks like `modifyOpInPlace` assert if the op in question has been replaced/erased already. Using `replaceAllUsesWith` uses `modifyOpInPlace` to update uses, so if any users have been replaced/erased we will hit the assertion.

I spent some time looking into the issue, for example the uses in this pass: https://github.com/llvm/circt/blob/e450388da7d6890d89b08e5277271d9afb6ad93b/lib/Dialect/Ibis/Transforms/IbisPortrefLowering.cpp#L86

I think we can probably do something like `replaceUsesWithIf`, with a condition to ensure we don't modify users that are already replaced/erased, but I wasn't able to figure out the right condition to avoid the assertion and still have the tests pass. In the LLVM bump PR, I've XFAIL'ed the Ibis tests that now hit this assertion.

@mortbopet @teqdruid is someone familiar with Ibis able to take a look at these?

mikeurbach, 2024-03-06:
I also asked on Discord if it would be reasonable for the DialectConversion infrastructure to handle this under the hood, and not modify ops that are already replaced/erased: https://discord.com/channels/636084430946959380/642426447167881246/1215040643722641528. I expect the answer is this is for us to handle as users of DialectConversion, but it would be easy to handle upstream if that is agreeable.

mikeurbach, 2024-03-07:
I've worked around this for now by following the recommendations on Discord, and opened the linked issue to fully resolve the unsafeness that I've intentionally introduced.

closed 2024-03-07 by mikeurbach

seldridge, 2024-03-07:
Thanks for working through this tedious issue, Mike!