llvm/circt #6808: [CI] LLVM Rebuild configuration suddenly broken?

maerhart, 2024-03-11:
In [my latest PR](https://github.com/llvm/circt/pull/6807), the CI decided to rebuild LLVM (no clue why the cache was suddenly gone). I neither bumped LLVM nor changed a workflow/config file. The build then failed very early on because the C and CXX compiler apparently changed after which it decides to restart the configuration step but fails because `DCMAKE_BUILD_TYPE` is not set anymore.  (tried to rerun the workflow once but same issue)

Anyone an idea what is going on here?

```
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= /usr/bin/gcc
CMAKE_CXX_COMPILER= /usr/bin/g++

[restarts configuration...]

CMake Error at CMakeLists.txt:97 (message):
  

  No build type selected.  You need to pass -DCMAKE_BUILD_TYPE=<type> in
  order to configure LLVM.

  Available options are:

    * -DCMAKE_BUILD_TYPE=Release - For an optimized build with no assertions or debug info.
    * -DCMAKE_BUILD_TYPE=Debug - For an unoptimized build with assertions and debug info.
    * -DCMAKE_BUILD_TYPE=RelWithDebInfo - For an optimized build with no assertions but with debug info.
    * -DCMAKE_BUILD_TYPE=MinSizeRel - For a build optimized for size instead of speed.
```
The run: https://github.com/llvm/circt/actions/runs/8229912405/job/22504074134

seldridge, 2024-03-12:
The MSFT runners were having some issues overnight. This correlates with that, though I don't know why it would manifest in the way the failure you're reporting did. 😕 The runners got fixed (thanks @teqdruid) and the CI queue drained. I restarted the failing run to see if things are better.

maerhart, 2024-03-12:
Thanks for restarting @seldridge, it seems to work again. Really interesting. Thanks for fixing the runners @teqdruid.

closed 2024-03-12 by maerhart