llvm/circt #6919: [FIRRTL] Not generate "automatic logic" in the output SystemVerilog file

SinaKarvandi, 2024-04-15:
Hi,
I'm using Chisel to generate SystemVerilog files for my design (https://github.com/HyperDbg/hwdbg).
For a simple FSM, FIRTOOL generated `automatic logic`, like the below example:

```
      automatic logic             _GEN = state == 3'h0;	// src/main/scala/hwdbg/interpreter/interpreter.scala:76:22, :97:19, :219:15
      automatic logic             _GEN_0 = state == 3'h1;	// src/main/scala/hwdbg/interpreter/interpreter.scala:76:22, :97:19, :121:17
      automatic logic             _GEN_1;	// src/main/scala/hwdbg/interpreter/interpreter.scala:97:19
      automatic logic             _GEN_2;	// src/main/scala/hwdbg/interpreter/interpreter.scala:97:19
      automatic logic             _GEN_3;	// src/main/scala/hwdbg/interpreter/interpreter.scala:97:19
      automatic logic             _GEN_4 = state == 3'h5;	// src/main/scala/hwdbg/interpreter/interpreter.scala:76:22, :97:19, :194:15
      automatic logic [31:0]      _GEN_5;	// src/main/scala/hwdbg/interpreter/interpreter.scala:88:40, :97:19
      automatic logic             _GEN_6;	// src/main/scala/hwdbg/interpreter/interpreter.scala:82:38, :97:19
      automatic logic [7:0][2:0]  _GEN_7 =
        {{state},
         {3'h0},
         {3'h6},
         {3'h5},
         {3'h4},
         {3'h3},
         {3'h2}
```

The problem is, none of my testing tools (e.g., iverilog) don't support such a statement as (`automatic logic`). Is there any parameter or way to force FIRTOOL not to use this statement in the generated code?

Thanks.

SinaKarvandi, 2024-04-15:
I add 
` --lowering-options=disallowLocalVariables`
to my options and it fixed the issue (it no longer generates `automatic logic`). 

Am I correctly addressing this issue? 🤔

uenoku, 2024-04-15:
Yes `disallowLocalVariables` is intended workaround for tools that don't support automatic logic.

closed 2024-04-15 by uenoku