llvm/circt #6922: [FIRRTL] ‘firtool’ crashes with an error when receives a ‘fir’ file with no main module

iagrigorov, 2024-04-16:
When one gives a ‘fir’ file with no main module as an input, ‘firtool’ crashes with the following message:
```
firtool: /home/grigorovia/circt/lib/Support/InstanceGraph.cpp:109: InstanceGraphNode *circt::igraph::InstanceGraph::lookup(StringAttr): Assertion `it != nodeMap.end() && "Module not in InstanceGraph!"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: ./build/bin/firtool /home/grigorovia/fir-tests/3.2.0/5_3-2.fir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool         0x0000560f4d2c0a97 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 39
1  firtool         0x0000560f4d2be4fe llvm::sys::RunSignalHandlers() + 238
2  firtool         0x0000560f4d2c13ef
3  libpthread.so.0 0x00007f4d21e48420
4  libc.so.6       0x00007f4d218db00b gsignal + 203
5  libc.so.6       0x00007f4d218ba859 abort + 299
6  libc.so.6       0x00007f4d218ba729
7  libc.so.6       0x00007f4d218cbfd6
8  firtool         0x0000560f4dcfa3d7 circt::igraph::InstanceGraph::lookup(mlir::StringAttr) + 215
9  firtool         0x0000560f4d937493 circt::firrtl::InstanceGraph::InstanceGraph(mlir::Operation*) + 307
10 firtool         0x0000560f4d3a2f4d
11 firtool         0x0000560f4d534ea2
12 firtool         0x0000560f4de527f6 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 694
13 firtool         0x0000560f4de53141 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 353
14 firtool         0x0000560f4de590b3
15 firtool         0x0000560f4de5496b mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) + 2299
16 firtool         0x0000560f4de5293b mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 1019
17 firtool         0x0000560f4de53141 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 353
18 firtool         0x0000560f4de558e5 mlir::PassManager::run(mlir::Operation*) + 933
19 firtool         0x0000560f4d248d39
20 firtool         0x0000560f4d247df9
21 firtool         0x0000560f4d24465e
22 firtool         0x0000560f4d242e9a main + 362
23 libc.so.6       0x00007f4d218bc083 __libc_start_main + 243
24 firtool         0x0000560f4d23f7fe _start + 46
Aborted (core dumped)
```

dtzSiFive, 2024-04-16:
This was fixed with 3cf4305d6408bf726eff7c2e1eea328225ccf76a .

What version of firtool are you using, can you use a newer one?  This was included as of 1.67.0 and works on main (and crashes as you report when reverting that commit).

closed 2024-04-16 by dtzSiFive

dtzSiFive, 2024-04-16:
Sorry -- please reopen or follow-up if there's anything further we can help with.  But I believe the underlying issue has been resolved (yours may not be though, let me know!).

iagrigorov, 2024-04-17:
It's OK. I'm just working with lower version of 'CIRCT'.