llvm/circt #6926: [FIRRTL] isX lowering doesn't work for bundles

dtzSiFive, 2024-04-17:
Consider this FIRRTL IR:
```
module {
  firrtl.circuit "IsX" {
    firrtl.module @IsX(in %x: !firrtl.bundle<a: uint<1>>, out %y: !firrtl.uint<1>) attributes {convention = #firrtl<convention scalarized>} {
      %0 = firrtl.int.isX %x : !firrtl.bundle<a: uint<1>>
      firrtl.strictconnect %y, %0 : !firrtl.uint<1>
    }
  }
}
```

Assertion failure in LowerToHW:

```
firtool: /home/will/src/sifive/circt/llvm/mlir/lib/IR/Types.cpp:126: unsigned int mlir::Type::getIntOrFloatBitWidth() const: Assertion `isIntOrFloat() && "only integers and floats have a bitwidth"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: ./build/bin/firtool isX.bundle.fir --mlir-disable-threading --mlir-print-ir-after-all
 #0 0x00005583e131a706 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/will/src/sifive/circt/llvm/llvm/lib/Support/Unix/Signals.inc:723:13
 #1 0x00005583e13187e0 llvm::sys::RunSignalHandlers() /home/will/src/sifive/circt/llvm/llvm/lib/Support/Signals.cpp:106:18
 #2 0x00005583e131ad9b SignalHandler(int) /home/will/src/sifive/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #3 0x00007f1378054eb0 __restore_rt (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x3deb0)
 #4 0x00007f13780a402c __pthread_kill_implementation (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x8d02c)
 #5 0x00007f1378054e06 gsignal (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x3de06)
 #6 0x00007f137803d8f5 abort (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x268f5)
 #7 0x00007f137803d819 _nl_load_domain.cold (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x26819)
 #8 0x00007f137804d686 (/nix/store/p3jshbwxiwifm1py0yq544fmdyy98j8a-glibc-2.38-27/lib/libc.so.6+0x36686)
 #9 0x00005583e1798e67 decltype(auto) llvm::cast<mlir::FloatType, mlir::Type>(mlir::Type const&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/Support/Casting.h:566:3
#10 0x00005583e1798e67 mlir::Type::getIntOrFloatBitWidth() const /home/will/src/sifive/circt/llvm/mlir/lib/IR/Types.cpp:129:10
#11 0x00005583e17b6da2 (anonymous namespace)::FIRRTLLowering::visitExpr(circt::firrtl::IsXIntrinsicOp) /home/will/src/sifive/circt/lib/Conversion/FIRRTLToHW/LowerToHW.cpp:3654:7
```

uenoku, 2024-04-17:
Can we make its verifier (or by ODS definition) reject non-integer type for isX?

darthscsi, 2024-04-30:
> Can we make its verifier (or by ODS definition) reject non-integer type for isX?

isX should work for bundles.

commit acbee5dc (2024-05-01): [FIRRTL] Fix isX lowering for aggregates

Oversight in lowering.  bitcast to an int to do the check.

Closes #6926
closed 2024-05-01 by darthscsi