llvm/circt #6983: Elaborate chisel type annotation from firtool to generate debug information for the Tywaves project

rameloni, 2024-05-03:
@fabianschuiki I created this issue to keep talking about the conversation we started here: https://github.com/chipsalliance/chisel/issues/4015#issuecomment-2077714172

I added attributes to debug variable to store the chisel type name and, eventually, the constructor parameters in the debug dialect (and maybe in `dbg.struct` and `dbg.array` as well).

However, I may need to create a dbg variable for each sub field, recursively, of aggregates. 
Is there a reason why values in `dbg.struct`/`dbg.array` are of Firrtl types rather than other `dbg.variable`s in `MaterializeDebugInfo`?

sequencer, 2024-05-03:
I think we can to expose the chisel type via compiler plugin like what we did in the source info. But what on earth is a chisel type? That’s the fundamental question(we need specification)

rameloni, 2024-05-08:
> I think we can to expose the chisel type via compiler plugin like what we did in the source info. 

I already managed to expose source level information from chisel to firrtl and firtool. In the meantime I also managed to read this information from firtool, processing it into proper debug variables and updating hgldd/emitting a new file format.

> But what on earth is a chisel type? That’s the fundamental question(we need specification)

Whit chisel type refer to the type of the scala variable in the source chisel code.
In the following example:

```scala
class MyBundle(size: Int) extends Bundle {
   val a = Bool()
   val b = SInt(size.W)
   val nested = new Bundle {
      val x = UInt(8.W)
   }
}
val inBundle  = IO(Input(new MyBundle(10)))
val wireBundle = Wire(new MyBundle(7))
val outBundle = IO(Output(new MyBundle(9)))) 
```
The types would be
| Variable          | Type                  |
| ----------------- | --------------------- |
| inBundle          | `IO[MyBundle(10)]`    |
| wireBundle        | `Wire[MyBundle(7)]`   |
| outBundle         | `IO[MyBundle(9)]`     |
| inBundle.a        | `IO[Bool]`            |
| inBundle.b        | `IO[SInt<10>]`        |
| inBundle.nested   | `IO[AnonymousBundle]` |
| inBundle.nested.x | `IO[UInt<8>]`         |
| ...               | ...                   |

Namely, I refer to anything represented in this table, including user-defined types:
https://raw.githubusercontent.com/chipsalliance/chisel/main/docs/src/images/type_hierarchy.svg?sanitize=true
This is basically the information I'm trying to emit in a file format similar to hgldd: https://github.com/chipsalliance/chisel/issues/4015#issuecomment-2066101886