llvm/circt #7040: [HW] hw.module.extern does not print non-identifier names correctly

Moxinilian, 2024-05-15:
```
hw.module.extern @foo(in %t "h ^hello" : i32)
```
is printed by circt-opt as
```
hw.module.extern @foo(in %h ^hello : i32)
```
which is not valid.

I may try to fix this if I have time but if anyone wants to beat me to it feel free.

thomasnormal, 2026-03-03:
Update: landed on `thomasnormal/circt` `main` in commit `4e6df27dd6`.

What landed:
- `lib/Dialect/HW/ModuleImplementation.cpp`
  - extern input port printer now checks SSA-identifier validity
  - if invalid, prints stable fallback SSA name (`%argN`) plus explicit string name (round-trip safe)
- `test/Dialect/HW/modules.mlir`
  - regression for weird extern input-port names (`"h ^hello"`)

Validation:
- `build_test/bin/llvm-lit -sv -j1 test/Dialect/HW/modules.mlir test/Dialect/HW/basic.mlir`
- result: `2 passed`

seldridge, 2026-03-06:
@thomasnormal, @thomasahle: Due to current CIRCT project policy, we are unable to accept patches from the `thomasnormal/circt` fork.

For others: please do not upstream any patches from the `thomasnormal/circt` fork. Additionally, recognize that any comments made by the @thomasnormal account appear to be from one or more AI agents.

For more information, please see this Discourse post: https://discourse.llvm.org/t/circt-project-policy-on-contributions-from-thomasnormal-circt-fork/90075