llvm/circt #7047: `-DMLIR_ENABLE_EXPENSIVE_PATTERN_API_CHECKS=ON` might have found some invalid API usages

7FM, 2024-05-16:
While the [documentation](https://mlir.llvm.org/getting_started/Debugging/#detecting-invalid-api-usage) states that there might be false positives, I think it might be worth a look into the individual test cases and patterns.
If there are no false positives, adding `-DMLIR_ENABLE_EXPENSIVE_PATTERN_API_CHECKS=ON` to the CI could be beneficial.
```
Failed Tests (32):
  CIRCT :: Conversion/LoopScheduleToCalyx/convert_pipeline.mlir
  CIRCT :: Conversion/SCFToCalyx/cider_source_location.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_func.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_simple.mlir
  CIRCT :: Conversion/SCFToCalyx/errors.mlir
  CIRCT :: Dialect/Arc/arc-canonicalizer.mlir
  CIRCT :: Dialect/Arc/canonicalizers.mlir
  CIRCT :: Dialect/Calyx/canonicalization.mlir
  CIRCT :: Dialect/Comb/canonicalization.mlir
  CIRCT :: Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/data-taps.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/invalid-interpretations.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/invalid-reg-pass.fir
  CIRCT :: Dialect/FIRRTL/SFCTests/mem-taps.fir
  CIRCT :: Dialect/FIRRTL/canonicalization.mlir
  CIRCT :: Dialect/FIRRTL/ref.mlir
  CIRCT :: Dialect/FIRRTL/simplify-mems.mlir
  CIRCT :: Dialect/HW/canonicalization.mlir
  CIRCT :: Dialect/HW/inline.mlir
  CIRCT :: Dialect/Handshake/canonicalization.mlir
  CIRCT :: Dialect/LLHD/Canonicalization/extract.mlir
  CIRCT :: firtool/async-reset.fir
  CIRCT :: firtool/chirrtl.fir
  CIRCT :: firtool/firtool.fir
  CIRCT :: firtool/import-ref.fir
  CIRCT :: firtool/lower-memories.fir
  CIRCT :: firtool/prefixMemory.fir
  CIRCT :: firtool/spec/refs/define.fir
  CIRCT :: firtool/spec/refs/nested_refproducer.fir
  CIRCT :: firtool/sv-attr.fir

Testing Time: 2.55s

Total Discovered Tests: 812
  Unsupported      :  12 (1.48%)
  Passed           : 762 (93.84%)
  Expectedly Failed:   6 (0.74%)
  Failed           :  32 (3.94%)
```

The observed errors are:
- `LLVM ERROR: operation finger print changed`
- `LLVM ERROR: pattern returned success but IR did not change`
- `LLVM ERROR: IR failed to verify after pattern application`

dtzSiFive, 2024-05-16:
Thanks for trying this and reporting!

I'm seeing 34 errors presently on my setup, here's a log but I'll start going through these... https://gist.github.com/dtzSiFive/cc4c9652739f455bb36e36490c98d940 .

Enabling debug prints for pattern application (not included above) helps indicate what pattern caused the bug.

dtzSiFive, 2024-05-17:
This is now down to 11 -- 10 once the Handshake PR lands.

Here's the test list, FWIW:

```
Failed Tests (11):
  CIRCT :: Conversion/LoopScheduleToCalyx/convert_pipeline.mlir
  CIRCT :: Conversion/SCFToCalyx/cider_source_location.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_controlflow.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_func.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_memory.mlir
  CIRCT :: Conversion/SCFToCalyx/convert_simple.mlir
  CIRCT :: Conversion/SCFToCalyx/errors.mlir
  CIRCT :: Dialect/Arc/arc-canonicalizer.mlir
  CIRCT :: Dialect/Arc/canonicalizers.mlir
  CIRCT :: Dialect/Calyx/canonicalization.mlir
  CIRCT :: Dialect/Handshake/canonicalization.mlir
```

uenoku, 2026-08-25:
This is now checked in nightly CI. At least we wouldn't get new failures 🫡