llvm/circt #7142: [HW] Incorrect SSA names for instance op.

uenoku, 2024-06-07:
Reported in [discord](https://discord.com/channels/636084430946959380/742572728787402763/1248378408216559686)

```bash
$ cat bar.mlir
hw.module.extern @foo( in %in_0 : i32, out "out0" : i1 )
hw.module @bar(out "out0" : i1) {
    %t1 = hw.constant -3 : i32
    %t2 = hw.instance "%103" @foo(in_0 : %t1 : i32) -> (out0 : i1)
    hw.output  %t2 : i1
}

$ circt-opt bar.mlir
  hw.module @bar(out out0 : i1) {
    %c-3_i32 = hw.constant -3 : i32
    %25103.out0 = hw.instance "%103" @foo(in_0: %c-3_i32: i32) -> (out0: i1)
    hw.output %25103.out0 : i1
  }

$ circt-opt bar.mlir | circt-opt
  hw.module @bar(out out0 : i1) {
    %c-3_i32 = hw.constant -3 : i32
    %25103.out0 = hw.instance "%103" @foo(in_0: %c-3_i32: i32) -> (out0: i1)
    hw.output %25103.out0 : i1
  }

<stdin>:5:12: error: expected three consecutive dots for an ellipsis
    %25103.out0 = hw.instance "%103" @foo(in_0: %c-3_i32: i32) -> (out0: i1)
``` 

h/t Steven Derrien

dtzSiFive, 2024-06-07:
I think MLIR's [`sanitizeIdentifier` method](https://github.com/llvm/llvm-project/blob/f9ae07b9e1509732be01fddefb529c6626c119a9/mlir/lib/IR/AsmPrinter.cpp#L997-L1040) has a bug where it fails to avoid leading digit, it checks for this in the name given but can still produce invalid identifier name if first character isn't valid (uses `llvm::utohexstr`) (which may produce a leading digit as we're seeing here).

We can workaround this by sanitizing ourselves but probably best to fix upstream.

dtzSiFive, 2024-06-07:
Sent quick fix upstream: https://github.com/llvm/llvm-project/pull/94795 .

uenoku, 2024-06-08:
Awesome, thanks!

dtzSiFive, 2024-06-11:
PR landed, this will be resolved with next LLVM bump.

dtzSiFive, 2024-07-01:
Bump is now included, example works.  Closing!