llvm/circt #7148: [HW] InstanceOp verifier failure

HahaLan97, 2024-06-10:
If I create a module with parameter as:
```
hw.module @test<param: i32>(in in0: !esi.channel<!hw.int<#hw.param.decl.ref<"param">>>, out: ...) { ... }
```
Then instantiate it using a !esi.channel, say channel0:
```
%outputs = hw.instance "test" @test<param: i32 = 1>(in0: channel0 : !esi.channel<i1>) -> (out: ...)
```
Then this error will show up:
```
'hw.instance' op operand type #0 must be '!esi.channel<!hw.int<#hw.param.decl.ref<"param">>>', but got '!esi.channel<i1>'
```
However, if I directly use `!hw.int<#hw.param.decl.ref<"param">>` instead of wrap this in other types like `!esi.channel`, it will work with no problems. I guess there is something missing in the verifier of InstanceOp, specifically in the `resolveParametricTypes` function.
Any help would be great!

HahaLan97, 2024-06-10:
I tried to add a Case as:
```
.Case<esi::ChannelType>(
          [&](esi::ChannelType channelType) -> FailureOr<Type> {
            auto innerType = channelType.getInner();
            if (!isa<hw::IntType>(innerType))
              return {channelType};
            auto elementType =
                evaluateParametricType(loc, parameters, innerType, emitErrors);
            if (failed(elementType))
              return {failure()};
            return {
                esi::ChannelType::get(channelType.getContext(), *elementType)};
          })
```
and add `CIRCTESI` in the link part in `CMakeLists.txt`. Then there is a cyclic dependency error because ESI depends on `CIRCTHW`. Could there be any workaround?

teqdruid, 2024-06-10:
Parameterized modules are one of the known weak points in CIRCT, so I generally recommend against using them unless you're willing to contribute a serious amount of code to CIRCT.

Unless something has changed, types are mostly opaque in MLIR. As a result, I think the best (or only) way to solve this is to introduce a type interface with an `evaluateParametricType` method. Every wrapper type would have to opt into it, but I don't see any other way.

HahaLan97, 2024-06-11:
> Parameterized modules are one of the known weak points in CIRCT, so I generally recommend against using them unless you're willing to contribute a serious amount of code to CIRCT.
>
> Unless something [...]

Thanks for the reply! I can feel it from the very first place when I started trying to build anything using parameters in CIRCT. Yes, you're right, I don't think there is any other way for this as well. Besides, MLIR doesn't have dependent types, so I guess I'll redesign my work to not use them to avoid any further problems. Later at some point, I may try to add the interface as you mentioned ;)

closed 2024-06-11 by HahaLan97