llvm/circt #7152: [LLHD] Windows compile error

teqdruid, 2024-06-11:
The CI build breakage has been covering up a legit Windows build break.

```
[3677/4261] Building CXX object tools\circt\lib\Conversion\LLHDToLLVM\CMakeFiles\obj.CIRCTLLHDToLLVM.dir\LLHDToLLVM.cpp.obj
FAILED: tools/circt/lib/Conversion/LLHDToLLVM/CMakeFiles/obj.CIRCTLLHDToLLVM.dir/LLHDToLLVM.cpp.obj 
C:\BuildTools\VC\Tools\MSVC\14.29.30133\bin\Hostx64\x64\cl.exe  /nologo /TP -DGTEST_HAS_RTTI=0 -DUNICODE -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_NONSTDC_NO_WARNINGS -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_GLIBCXX_ASSERTIONS -D_HAS_EXCEPTIONS=0 -D_SCL_SECURE_NO_DEPRECATE -D_SCL_SECURE_NO_WARNINGS -D_UNICODE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -IC:\__w\1\b\build\tools\circt\lib\Conversion\LLHDToLLVM -IC:\__w\1\s\CIRCT\circt\lib\Conversion\LLHDToLLVM -IC:\__w\1\b\build\include -IC:\__w\1\s\CIRCT\circt\llvm\llvm\include -IC:\__w\1\s\CIRCT\circt\include -IC:\__w\1\b\build\tools\circt\include -IC:\vcpkg\installed\x64-windows\include -external:IC:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include -external:IC:\__w\1\b\build\tools\mlir\include -external:W0 /DWIN32 /D_WINDOWS   /Zc:inline /Zc:preprocessor /Zc:__cplusplus /Oi /bigobj /permissive- /W4 -wd4141 -wd4146 -wd4244 -wd4267 -wd4291 -wd4351 -wd4456 -wd4457 -wd4458 -wd4459 -wd4503 -wd4624 -wd4722 -wd4100 -wd4127 -wd4512 -wd4505 -wd4610 -wd4510 -wd4702 -wd4245 -wd4706 -wd4310 -wd4701 -wd4703 -wd4389 -wd4611 -wd4805 -wd4204 -wd4577 -wd4091 -wd4592 -wd4319 -wd4709 -wd5105 -wd4324 -w14062 -we4238 /Gw /O2 /Ob2  -std:c++17 -MD  /EHs-c- /GR- -UNDEBUG /showIncludes /Fotools\circt\lib\Conversion\LLHDToLLVM\CMakeFiles\obj.CIRCTLLHDToLLVM.dir\LLHDToLLVM.cpp.obj /Fdtools\circt\lib\Conversion\LLHDToLLVM\CMakeFiles\obj.CIRCTLLHDToLLVM.dir\ /FS -c C:\__w\1\s\CIRCT\circt\lib\Conversion\LLHDToLLVM\LLHDToLLVM.cpp
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Support/TypeID.h(193): error C2338: TypeID::get<> requires the complete definition of `T`
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Support/TypeID.h(192): note: while compiling class template member function 'mlir::TypeID mlir::detail::TypeIDResolver<T,void>::resolveTypeID(void)'
        with
        [
            T=circt::hw::HWEmittableModuleLike
        ]
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Support/TypeID.h(233): note: see reference to function template instantiation 'mlir::TypeID mlir::detail::TypeIDResolver<T,void>::resolveTypeID(void)' being compiled
        with
        [
            T=circt::hw::HWEmittableModuleLike
        ]
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Support/TypeID.h(233): note: see reference to class template instantiation 'mlir::detail::TypeIDResolver<T,void>' being compiled
        with
        [
            T=circt::hw::HWEmittableModuleLike
        ]
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/IR/OperationSupport.h(375): note: see reference to function template instantiation 'mlir::TypeID mlir::TypeID::get<T>(void)' being compiled
        with
        [
            T=circt::hw::HWEmittableModuleLike
        ]
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Pass/Pass.h(440): note: see reference to function template instantiation 'bool mlir::OperationName::hasInterface<InterfaceT>(void) const' being compiled
        with
        [
            InterfaceT=circt::hw::HWEmittableModuleLike
        ]
C:\__w\1\s\CIRCT\circt\llvm\llvm\..\mlir\include\mlir/Pass/Pass.h(439): note: while compiling class template member function 'bool mlir::InterfacePass<circt::hw::HWEmittableModuleLike>::canScheduleOn(mlir::RegisteredOperationName) const'
C:\__w\1\b\build\tools\circt\include\circt/Conversion/Passes.h.inc(5961): note: see reference to class template instantiation 'mlir::InterfacePass<circt::hw::HWEmittableModuleLike>' being compiled
C:\__w\1\b\build\tools\circt\include\circt/Conversion/Passes.h.inc(6009): note: see reference to class template instantiation 'circt::PrepareForEmissionBase<DerivedT>' being compiled
```

teqdruid, 2024-06-11:
This isn't just LLHD... the same error occurs in several other places. See the attached log.
[wincompile.txt](https://github.com/user-attachments/files/15780655/wincompile.txt)

teqdruid, 2024-06-11:
@uenoku this legit build break does appear to be related to `HWEmittableModuleLike`. It was just bad luck that the image issue pop'ed at exactly the same time.

uenoku, 2024-06-11:
Got it, thank you for letting me know!

uenoku, 2024-06-11:
Sorry probably some cmake dependencies are missing but I cannot test locally and CI still fails due to the image issue (even when I added a macro workaround https://github.com/llvm/circt/actions/runs/9465994254). I'll fix ASAP once the image is fixed.

dtzSiFive, 2024-06-12:
Image appears fixed (the new one finally deployed), so I restarted the latest Windows CI job and it's churning along: https://github.com/llvm/circt/actions/runs/9471333444/job/26148886537 .

teqdruid, 2024-06-12:
That compile survived the image issue but failed with this issue (as expected).

uenoku, 2024-06-13:
Fixed by https://github.com/llvm/circt/commit/2876be2297363ab29db74ae010be7641252a55bf

closed 2024-06-13 by uenoku