llvm/circt #7188: [CMake] Installation fails if SLANG_ENABLE_FRONTEND is enabled

cepheus69, 2024-06-16:
While I was trying to generate installation of CIRCT with enabled SLANG_ENABLE_FRONTEND option. The installation is failed. I went through the file `cmake_install.cmake`  and found snippets that should be relevant to this question.

```CMake
if(CMAKE_INSTALL_COMPONENT STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include" TYPE DIRECTORY FILES "/home/ursa/Projects/circt/build/_deps/fmt-src/include/fmt")
endif()
```
It looks like an extra piece of error code was generated when installing the header files for the fmt library. The wrong code tries to find the fmt library headers in circt's include path.
```CMake
if(CMAKE_INSTALL_COMPONENT STREQUAL "Unspecified" OR NOT CMAKE_INSTALL_COMPONENT)
  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include" TYPE FILE FILES
    "/home/ursa/Projects/circt/include/fmt/args.h"
    "/home/ursa/Projects/circt/include/fmt/chrono.h"
    "/home/ursa/Projects/circt/include/fmt/color.h"
    "/home/ursa/Projects/circt/include/fmt/compile.h"
    "/home/ursa/Projects/circt/include/fmt/core.h"
    "/home/ursa/Projects/circt/include/fmt/format.h"
    "/home/ursa/Projects/circt/include/fmt/format-inl.h"
    "/home/ursa/Projects/circt/include/fmt/os.h"
    "/home/ursa/Projects/circt/include/fmt/ostream.h"
    "/home/ursa/Projects/circt/include/fmt/printf.h"
    "/home/ursa/Projects/circt/include/fmt/ranges.h"
    "/home/ursa/Projects/circt/include/fmt/std.h"
    "/home/ursa/Projects/circt/include/fmt/xchar.h"
    )
endif()
```

cepheus69, 2024-06-16:
If remove the second piece of code in `cmake_install.cmake`. Then the installation can proceed normally.

cepheus69, 2024-06-16:
I'm trying to fix it. So are there any other details I'm missing? @fabianschuiki

SpriteOvO, 2024-06-16:
Can you provide the error message from CMake and commands you executed? Do you have the `fmt` package installed on your system?

cepheus69, 2024-06-16:
> Can you provide the error message from CMake and commands you executed? Do you have the `fmt` package installed on your system?

Sure! My building commands are there:
```bash
  cmake -G Ninja .. \
      -DCMAKE_BUILD_TYPE=$BUILD_TYPE \
      -DCMAKE_C_COMPILER=$CC \
      -DCMAKE_CXX_COMPILER=$CXX \
      -DLLVM_USE_LINKER=lld \
      -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
      -DCIRCT_SLANG_FRONTEND_ENABLED=ON \
      -DCMAKE_INSTALL_PREFIX=../../$INSTALL_DIR_1 \
      -DCMAKE_INSTALL_DOCDIR=~/Documents/CirctDoc \
      -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir \
      -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm \
      -DLLVM_ENABLE_ASSERTIONS=ON 
ninja
ninja check-circt
ninja check-circt-integration 
ninja install
```
BuildType is **Debug**, set **clang 18.1.7** as building compiler.
The log of installation:
```bash
Total Discovered Tests: 815
  Unsupported      :   5 (0.61%)
  Passed           : 804 (98.65%)
  Expectedly Failed:   6 (0.74%)
+ '[' 0 '!=' 0 ']'
+ ninja check-circt-integration
[1/2] Running the CIRCT integration tests

Testing Time: 27.44s

Total Discovered Tests: 97
  Unsupported      : 61 (62.89%)
  Passed           : 35 (36.08%)
  Expectedly Failed:  1 (1.03%)
+ '[' 0 '!=' 0 ']'
+ install
+ cd /Users/xubuyun/Projects/circt/build
+ ninja install
[0/1] Install the project...
-- Install configuration: "DEBUG"
-- Installing: /Users/xubuyun/Projects/install-circt/lib/libsvlang.a
-- Installing: /Users/xubuyun/Projects/install-circt/lib/libfmtd.a
CMake Error at cmake_install.cmake:57 (file):
  file INSTALL cannot find
  "/Users/xubuyun/Projects/circt/include/fmt/args.h": No such file or
  directory.

FAILED: CMakeFiles/install.util 
cd /Users/xubuyun/Projects/circt/build && /opt/homebrew/Cellar/cmake/3.29.5/bin/cmake -P cmake_install.cmake
ninja: build stopped: subcommand failed.
+ cd /Users/xubuyun/Projects/circt/llvm/build
+ ninja install
[0/1] Re-running CMake...
/bin/sh: /opt/homebrew/Cellar/cmake/3.29.3/bin/cmake: No such file or directory
FAILED: build.ninja 
/opt/homebrew/Cellar/cmake/3.29.3/bin/cmake --regenerate-during-build -S/Users/xubuyun/Projects/circt/llvm/llvm -B/Users/xubuyun/Projects/circt/llvm/build
ninja: error: rebuilding 'build.ninja': subcommand failed
```

fabianschuiki, 2024-06-17:
I wish we could somehow get Slang to not build its third-party libraries in a way that forces us to install them :cry: Ideally we always link slang into a self-contained static library, together with all its dependencies, and then link everything into the `CIRCTImportVerilog` conversion library. Ideally at that point we wouldn't have to install any of the slang libraries or headers, since none of the slang implementation details are visible in public headers of `CIRCTImportVerilog`. From the outside it looks just like a source file into MLIR translation. Maybe we'll have to create a fork of Slang 3.0 and fix a few of these things?

teqdruid, 2024-06-18:
@fabianschuiki How's it going getting slang bumped to 4 or 5? They might have simplified the required libraries.

cepheus69, 2024-06-18:
> I wish we could somehow get Slang to not build its third-party libraries in a way that forces us to install them 😢 Ideally we always link slang into a self-contained static library, together with all its dependencies, and then link everything into the `CIRCTImportVerilog` conversion library. Ideally at that point we wouldn't have to install any of the slang libraries or headers, since none of the slang implementation details are visible in public headers of `CIRCTImportVerilog`. From the outside it looks just like a source file into MLIR translation. Maybe we'll have to create a fork of Slang 3.0 and fix a few of these things?

Yes. It's quite necessary to fix this, and it's an inescapable problem for pushing `circt-verilog` into chipsalliance's sv-tests repository as the new resident test item

cepheus69, 2024-06-18:
> @fabianschuiki How's it going getting slang bumped to 4 or 5? They might have simplified the required libraries.

From Slang v4.0, the author Mike uplifted the minimal language requirements from CXX17 to CXX20, however, the MLIR / CIRCT project mainline still adopts CXX17. Introducing slang 4.0 or above with C++20 while Circt still uses C++17 is still risky.

cepheus69, 2024-06-18:
It needs more work to ensure no conflicts between these two projects.

teqdruid, 2024-06-18:
I seem to recall spending a bunch of time getting a compiler which supports c++20 in the CI images. IIRC, we said at the time that Slang support would be optional and require a c++20 compiler.

teqdruid, 2024-06-19:
*(chanting)* do the bump! do the bump! do the bump!

cepheus69, 2024-08-06:
The problem was fixed, so close!

closed 2024-08-06 by cepheus69