llvm/circt #7194: [FIRRTL] SpecializeLayers failed in CI

fzi-hielscher, 2024-06-17:
I ran into a spurious CI failure after merging 9afdd4d069df511faa629ffde5d037c7e2fcf6a4 :
https://github.com/llvm/circt/actions/runs/9549355360/attempts/1

The log shows a crash in the test of `SpecializeLayers`:
```
2024-06-17T14:30:13.4526767Z ******************** TEST 'CIRCT :: Dialect/FIRRTL/specialize-layers.mlir' FAILED ********************
2024-06-17T14:30:13.4527674Z Exit Code: 2
2024-06-17T14:30:13.4527913Z 
2024-06-17T14:30:13.4528082Z Command Output (stderr):
2024-06-17T14:30:13.4528567Z --
2024-06-17T14:30:13.4531389Z RUN: at line 1: /__w/circt/circt/build/bin/circt-opt --pass-pipeline='builtin.module(firrtl.circuit(firrtl-specialize-layers))' /__w/circt/circt/test/Dialect/FIRRTL/specialize-layers.mlir | /__w/circt/circt/llvm/install/bin/FileCheck /__w/circt/circt/test/Dialect/FIRRTL/specialize-layers.mlir
2024-06-17T14:30:13.4535457Z + /__w/circt/circt/build/bin/circt-opt '--pass-pipeline=builtin.module(firrtl.circuit(firrtl-specialize-layers))' /__w/circt/circt/test/Dialect/FIRRTL/specialize-layers.mlir
2024-06-17T14:30:13.4538071Z + /__w/circt/circt/llvm/install/bin/FileCheck /__w/circt/circt/test/Dialect/FIRRTL/specialize-layers.mlir
2024-06-17T14:30:13.4539662Z PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
2024-06-17T14:30:13.4541513Z  #0 0x00007f970605dd0f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/__w/circt/circt/llvm/install/lib/libLLVMSupport.so.19.0git+0x25dd0f)
2024-06-17T14:30:13.4543015Z  #1 0x00007f970605b234 SignalHandler(int) Signals.cpp:0:0
2024-06-17T14:30:13.4544183Z  #2 0x00007f9707dd1420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
2024-06-17T14:30:13.4545524Z  #3 0x0000560a41f05b05 (anonymous namespace)::SpecializeLayers::operator()() SpecializeLayers.cpp:0:0
2024-06-17T14:30:13.4547089Z  #4 0x0000560a41f07bc5 (anonymous namespace)::SpecializeLayersPass::runOnOperation() SpecializeLayers.cpp:0:0
2024-06-17T14:30:13.4549367Z  #5 0x00007f9706eff071 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/__w/circt/circt/llvm/install/lib/libMLIRPass.so.19.0git+0x3b071)
2024-06-17T14:30:13.4553159Z  #6 0x00007f9706eff55a mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/__w/circt/circt/llvm/install/lib/libMLIRPass.so.19.0git+0x3b55a)
2024-06-17T14:30:13.4557661Z  #7 0x00007f9706effeec mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&)::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const Pass.cpp:0:0
2024-06-17T14:30:13.4570445Z  #8 0x00007f9706f000df std::_Function_handler<void (), mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&)&>(mlir::MLIRContext*, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&)&)::'lambda'()>::_M_invoke(std::_Any_data const&) Pass.cpp:0:0
2024-06-17T14:30:13.4583228Z  #9 0x0000560a4192ae62 std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<std::function<void ()>>>, void>>::_M_invoke(std::_Any_data const&) (/__w/circt/circt/build/bin/circt-opt+0xfc8e62)
2024-06-17T14:30:13.4587370Z #10 0x0000560a4192c2bd std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) ExportVerilog.cpp:0:0
2024-06-17T14:30:13.4589596Z #11 0x00007f9707dce4df __pthread_once_slow /build/glibc-wuryBv/glibc-2.31/nptl/pthread_once.c:118:7
2024-06-17T14:30:13.4592285Z #12 0x0000560a4192bac8 std::__future_base::_Deferred_state<std::thread::_Invoker<std::tuple<std::function<void ()>>>, void>::_M_complete_async() (/__w/circt/circt/build/bin/circt-opt+0xfc9ac8)
2024-06-17T14:30:13.4595644Z #13 0x0000560a4192e002 std::_Function_handler<void (), std::shared_future<void> llvm::ThreadPoolInterface::asyncImpl<void>(std::function<void ()>, llvm::ThreadPoolTaskGroup*)::'lambda'()>::_M_invoke(std::_Any_data const&) (/__w/circt/circt/build/bin/circt-opt+0xfcc002)
2024-06-17T14:30:13.4598617Z #14 0x00007f9705fe32db llvm::StdThreadPool::processTasks(llvm::ThreadPoolTaskGroup*) (/__w/circt/circt/llvm/install/lib/libLLVMSupport.so.19.0git+0x1e32db)
2024-06-17T14:30:13.4600964Z #15 0x00007f9705fe4119 void* llvm::thread::ThreadProxy<std::tuple<llvm::StdThreadPool::grow(int)::'lambda'()>>(void*) ThreadPool.cpp:0:0
2024-06-17T14:30:13.4602762Z #16 0x00007f9707dc5609 start_thread /build/glibc-wuryBv/glibc-2.31/nptl/pthread_create.c:478:7
2024-06-17T14:30:13.4604437Z #17 0x00007f9705a9a353 __clone /build/glibc-wuryBv/glibc-2.31/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:97:0
2024-06-17T14:30:13.4605610Z FileCheck error: '<stdin>' is empty.
2024-06-17T14:30:13.4607086Z FileCheck command line:  /__w/circt/circt/llvm/install/bin/FileCheck /__w/circt/circt/test/Dialect/FIRRTL/specialize-layers.mlir
2024-06-17T14:30:13.4608260Z 
2024-06-17T14:30:13.4608435Z --
2024-06-17T14:30:13.4608644Z 
2024-06-17T14:30:13.4608781Z ********************
```

I have not been able to reproduce it, neither locally nor in the CI, so I _guess_ it is caused by a race condition and I  _guess_ it is not actually related to the commit it failed on. But I also don't really have a clue on what is happening here.

youngar, 2024-06-18:
We were looking for bugs in specialize layers and found [this](https://github.com/llvm/circt/pull/7200) issue, but unfortunately I don't think its related to this issue, as the tests did not trigger it.  I still have not been able to reproduce;  We tried running with ASan, TSan, and Valgrind so far, and nothing has come up.

fzi-hielscher, 2024-06-18:
It crashed again. This time on an unrelated test run of a local macOS/clang debug build based on 6cbca83fe5123e04b323f1f37ba6f733ee4dadc2 .
I've attached the stack trace. It occurred in the `getOps<hw::HierPathOp>()` call at SpecializeLayers.cpp:656. Maybe this helps a bit pinning it down, @youngar.
[stacktrace.txt](https://github.com/user-attachments/files/15892904/stacktrace.txt)

fzi-hielscher, 2024-06-18:
As I've said, I don't actually know what is happening here, so I may well be talking rubbish:
I think it is strange that the crash appears to be due to a race condition, but it did not seem to occur within a parallel section of the pass. I then noticed that there are two nearly identical firrtl.circuit operations in the `specialize-layers.mlir` both called "HierPathDelete". Is this intentional?
So - as a strange thought - is it possible the paths are somehow cross referenced between the circuit ops?

fzi-hielscher, 2024-06-19:
So... giving this a proper look the explanation seems to be more mundane. I can somewhat reliably reproduce the crash with this snippet:

```
firrtl.circuit "DisableLayerA" attributes {
    disable_layers = [@A]
  } {
  firrtl.layer @B bind { }
  firrtl.layer @A bind { }  
  firrtl.extmodule @Test0() attributes {layers = [@A]}
  firrtl.extmodule @DisableLayerA()
}
```  

Note that the second and third operation in the circuit will be processed, and deleted, in concurrent invocations of `specializeTopLevelOp`. Since operations are nodes in a doubly linked list, this does not sound like a good idea to me.
If I swap around the layer operations, thus creating a stable node between the deleted operations, it does not crash.

youngar, 2024-06-20:
Wow thank you so much for tracking down this issue!

closed 2024-06-20 by youngar