llvm/circt #7197: [CMake] Both local install and compiled release usages failing

Muxianesty, 2024-06-18:
# Context

Platform: Ubuntu 22.04
Architecture: x86_64
CIRCT commit: tag `firtool-1.72.0` and newer

# Issue description
I've been trying for a while to use CIRCT in my projects, and for all this time I could not understand why there are issues with integrating both pre-compiled and locally built CIRCT.

With the locally built CIRCT (even with the simplest compiling procedure, described here https://github.com/llvm/circt?tab=readme-ov-file#setting-this-up ), somehow include-directories become "mixed" (see example below), while there is no such issue with pre-compiled CIRCT.

In this case, strings like `-I/home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/home/user/circt72/include` appear, which make an ill-formed include directory path, making it look like FIRRTL dialect's files couldn't be found.

`[25/85] Building CXX object src/hls/model/dfcir/lib/dfcir/CMakeFiles/obj.MLIRDFCIR.dir/conversions/DFCIRLinearSchedulerPass.cpp.o
FAILED: src/hls/model/dfcir/lib/dfcir/CMakeFiles/obj.MLIRDFCIR.dir/conversions/DFCIRLinearSchedulerPass.cpp.o 
/usr/bin/c++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D_GLIBCXX_ASSERTIONS -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/user/circt72/llvm/mlir/include -I/home/user/circt72/llvm/build/tools/mlir/include -I/home/user/utopia-hls/src/hls/model/dfcir/include -I/home/user/utopia-hls/build/src/hls/model/dfcir/include -I/home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/home/user/circt72/llvm/llvm/include -I/home/user/circt72/llvm/build/include -I/home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/home/user/circt72/llvm/mlir/include -I/home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/home/user/circt72/include -I/home/user/circt72/build/include -I/home/user/circt72/llvm/llvm/include -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -fno-lifetime-dse -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-nonnull -Wno-class-memaccess -Wno-redundant-move -Wno-pessimizing-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wsuggest-override -Wno-comment -Wno-misleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections  -fno-exceptions -funwind-tables -fno-rtti -UNDEBUG -g3 -Wall -Wignored-qualifiers -DYYERROR_VERBOSE -std=gnu++17 -MD -MT src/hls/model/dfcir/lib/dfcir/CMakeFiles/obj.MLIRDFCIR.dir/conversions/DFCIRLinearSchedulerPass.cpp.o -MF src/hls/model/dfcir/lib/dfcir/CMakeFiles/obj.MLIRDFCIR.dir/conversions/DFCIRLinearSchedulerPass.cpp.o.d -o src/hls/model/dfcir/lib/dfcir/CMakeFiles/obj.MLIRDFCIR.dir/conversions/DFCIRLinearSchedulerPass.cpp.o -c /home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/conversions/DFCIRLinearSchedulerPass.cpp
In file included from /home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/conversions/DFCIRLinearSchedulerPass.cpp:2:
/home/user/utopia-hls/src/hls/model/dfcir/include/dfcir/conversions/DFCIRPassesUtils.h:4:10: fatal error: circt/Dialect/FIRRTL/FIRRTLDialect.h: No such file or directory
    4 | #include "circt/Dialect/FIRRTL/FIRRTLDialect.h"
`
With the pre-built CIRCT, somehow it is required to manually edit `MLIRTargets.cmake`, because targets belonging to CIRCT appear in MLIR configuration files, which of course fails the compilation with the messages like the one below.

`CMake Error at /home/user/Downloads/firtool-1.72.0/lib/cmake/circt/CIRCTConfig.cmake:10 (find_package):
Found package configuration file:
  /home/user/Downloads/firtool-1.72.0/lib/cmake/mlir/MLIRConfig.cmake
  but it set MLIR_FOUND to FALSE so package "MLIR" is considered to be NOT
  FOUND.  Reason given by package:
  The following imported targets are referenced, but are missing:
  CIRCTAffineToLoopSchedule CIRCTArcToLLVM CIRCTCalyxToFSM CIRCTCalyxToHW
  CIRCTCalyxNative CIRCTCombToArith CIRCTCombToLLVM CIRCTCombToSMT
  CIRCTConvertToArcs CIRCTDCToHW CIRCTExportChiselInterface
  CIRCTExportVerilog CIRCTFIRRTLToHW CIRCTFSMToSV CIRCTHandshakeToDC
  CIRCTHandshakeToHW CIRCTHWArithToHW CIRCTHWToLLHD CIRCTHWToLLVM
  CIRCTHWToBTOR2 CIRCTHWToSMT CIRCTHWToSV CIRCTHWToSystemC CIRCTLLHDToLLVM
  CIRCTLoopScheduleToCalyx CIRCTMooreToCore CIRCTPipelineToHW CIRCTSCFToCalyx
  CIRCTSeqToSV CIRCTSimToSV CIRCTCFToHandshake CIRCTVerifToSMT CIRCTVerifToSV
  CIRCTExportFIRRTL CIRCTComb CIRCTCombTransforms CIRCTDebug CIRCTESI
  CIRCTFIRRTL CIRCTImportFIRFile CIRCTMSFT CIRCTMSFTTransforms CIRCTHW
  CIRCTLLHD CIRCTMoore CIRCTOM CIRCTOMEvaluator CIRCTSeq CIRCTSeqTransforms
  CIRCTSV CIRCTSVTransforms CIRCTFSM CIRCTFSMTransforms CIRCTHandshake
  CIRCTHandshakeTransforms CIRCTHWArith CIRCTVerif CIRCTLTL CIRCTEmit
  CIRCTFirtool
Call Stack (most recent call first):
  CMakeLists.txt:55 (find_package)
`
I have seen a similar issue (https://github.com/llvm/circt/issues/6418), but nothing has changed since then.

Am I doing something wrong or is it a widespread issue?

uenoku, 2024-06-19:
Could you trying using following cmake configuration? Internally we used this and it is working. 
```
// No LLVM
find_package(MLIR CONFIG)
find_package(CIRCT CONFIG)
```

Muxianesty, 2024-06-19:
> Could you trying using following cmake configuration? Internally we used this and it is working.
> 
> ```
> // No LLVM
> find_package(MLIR CONFIG)
> find_package(CIRCT CONFIG)
> ```

No, strings like `-I/home/user/utopia-hls/src/hls/model/dfcir/lib/dfcir/home/user/circt72/include` appear; paths are still broken.