llvm/circt #7212: [ImportVerilog] Upgrade slang to slang 5

teqdruid, 2024-06-19:
When slang was being added as a dep, the CI images couldn't handle C++20. They were very quickly upgraded to make them optionally use a newer clang version which supported C++20. The idea was to use slang 5 (which now supports compiling without RTTI and exceptions (IIRC) when a C++20 compiler was detected.

This hasn't happened yet.

teqdruid, 2024-07-01:
@fabianschuiki How's this coming? (poke)

astrophysik, 2024-10-30:
Hi, is there any work in progress to update the version of Slang used in CIRCT?

fabianschuiki, 2024-10-30:
Not yet, but we should definitely try to do it! :smiley: Since Slang is an optional feature of CIRCT, I don't think we have to care about the C++20 requirement too much. If users want Slang they'll have to have a C++20 compliant compiler. As long as our CI can run all the ImportVerilog tests, we're good :partying_face:1

teqdruid, 2024-10-31:
@fabianschuiki I spent a bunch of time getting the CI images to support C++20 when you first started (and I'm pretty sure I tested them with Slang 5) so they should be good.

hovind, 2024-10-31:
> Not yet, but we should definitely try to do it! 😃 Since Slang is an optional feature of CIRCT, I don't think we have to care about the C++20 requirement too much. If users want Slang they'll have to have a C++20 compliant compiler. As long as our CI can run all the ImportVerilog tests, we're good 🥳
> 
> If you would like to give this a shot, try changing this line in CMakeLists.txt and see what kind of errors appear: https://github.com/llvm/circt/blob/main/CMakeLists.txt#L560

I did some initial testing at https://github.com/llvm/circt/commit/37597be2ced28b17221416a91ff45fbccec16d3f, and the required changes to compile against the current master seem pretty minor. I haven't checked `v5.0` yet.

`ninja -C build/ check-circt` currently outputs
```
Failed Tests (1):
  CIRCT-Unit :: Support/./CIRCTSupportTests/1/26
```
I'll take a look at that when I get time, unless somebody else beats me to it!

The most annoying thing I ran into was that they (as far as I can tell) require the input buffers to be zero terminated https://github.com/MikePopoloski/slang/blob/bad0d6ba3dec20355b0156c535bbddf2f926ab37/source/parsing/Lexer.cpp#L49, which means I failed to remove the extra copying described [here](https://github.com/llvm/circt/blob/afd11bbca0a7dd0c3eae1b90ae4d53e51e60df66/lib/Conversion/ImportVerilog/ImportVerilog.cpp#L197).

fabianschuiki, 2024-10-31:
That sounds very promising. Bumping this to v5 would certainly make @teqdruid very happy 😁

hovind, 2024-11-01:
Any reason to prefer `v5.0`? If we want the `llvm::SourceMgr` to manage the input buffers, I think we'll need https://github.com/MikePopoloski/slang/pull/1164, which has been merged after `v7.0`. I pushed an updated draft to https://github.com/llvm/circt/compare/main...hovind:circt:dev/hovind/slang-master.

fabianschuiki, 2024-11-01:
I'd be very happy with version 7 as well! :+1:

teqdruid, 2024-11-02:
> You are receiving this email because you authored the thread.

Ideally, we'd keep up with version bumps -- always stick to the latest.

astrophysik, 2024-11-15:
Is anyone doing this task? If not, I can do it

fabianschuiki, 2024-11-15:
The shared/static builds are definitely an adventure :see_no_evil: I was originally hoping that we could statically link Slang into ImportVerilog regardless of whether CIRCT was built with dynamic or static libraries. No idea if that's reasonable or even achievable.

hovind, 2024-11-18:
> The shared/static builds are definitely an adventure 🙈 I was originally hoping that we could statically link Slang into ImportVerilog regardless of whether CIRCT was built with dynamic or static libraries. No idea if that's reasonable or even achievable.

As far as I understand, that is how it works today thanks to the `set(BUILD_SHARED_LIBS OFF)` before `FetchContent_MakeAvailable(slang)`.

What I have been struggling with is that when `llvm_add_library()` decides to build a [object library](https://cmake.org/cmake/help/latest/command/add_library.html#object-libraries) [here](https://github.com/llvm/llvm-project/blob/b7d635ed30da49cc32b5b46d00e67ecc3ff9522f/llvm/cmake/modules/AddLLVM.cmake#L565), `CMake` generates the following `build.ninja` target:
```
build lib/Conversion/ImportVerilog/CMakeFiles/obj.CIRCTImportVerilog.dir/ImportVerilog.cpp.o: CXX_COMPILER__obj.2eCIRCTImportVerilog_unscanned_Debug /home/oystein/build/circt/lib/Conversion/ImportVerilog/ImportVerilog.cpp || cmake_object_order_depends_target_obj.CIRCTImportVerilog
  DEFINES = -DDEBUG -DGTEST_HAS_RTTI=0 -D_DEBUG -D_GLIBCXX_ASSERTIONS -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
  DEP_FILE = lib/Conversion/ImportVerilog/CMakeFiles/obj.CIRCTImportVerilog.dir/ImportVerilog.cpp.o.d
  FLAGS = -fno-exceptions -fno-rtti -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -g -std=gnu++20  -fno-exceptions -funwind-tables -fno-rtti -Wno-non-virtual-dtor -Wno-ctad-maybe-unsupported -Wno-covered-switch-default
  INCLUDES = -I/home/oystein/build/circt/llvm/llvm/include -I/home/oystein/build/circt/llvm/build/include -I/home/oystein/build/circt/llvm/mlir/include -I/home/oystein/build/circt/llvm/build/tools/mlir/include -I/home/oystein/build/circt/include -I/home/oystein/build/circt/build/include -I/home/oystein/build/circt/build/_deps/slang-src/source/../include -I/home/oystein/build/circt/build/_deps/slang-build/source -I/home/oystein/build/circt/build/_deps/slang-src/source/../external -I/home/oystein/build/circt/build/_deps/fmt-src/include
  OBJECT_DIR = lib/Conversion/ImportVerilog/CMakeFiles/obj.CIRCTImportVerilog.dir
  OBJECT_FILE_DIR = lib/Conversion/ImportVerilog/CMakeFiles/obj.CIRCTImportVerilog.dir
```
compared to the target when an object library is not generated:
```
build lib/Conversion/ImportVerilog/CMakeFiles/CIRCTImportVerilog.dir/ImportVerilog.cpp.o: CXX_COMPILER__CIRCTImportVerilog_unscanned_Debug /home/oystein/build/circt/lib/Conversion/ImportVerilog/ImportVerilog.cpp || cmake_object_order_depends_target_CIRCTImportVerilog
  DEFINES = -DDEBUG -DGTEST_HAS_RTTI=0 -DSLANG_BOOST_SINGLE_HEADER -DSLANG_STATIC_DEFINE -D_DEBUG -D_GLIBCXX_ASSERTIONS -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
  DEP_FILE = lib/Conversion/ImportVerilog/CMakeFiles/CIRCTImportVerilog.dir/ImportVerilog.cpp.o.d
  FLAGS = -fno-exceptions -fno-rtti -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -g -std=gnu++20 -fPIC  -fno-exceptions -funwind-tables -fno-rtti -Wno-non-virtual-dtor -Wno-ctad-maybe-unsupported -Wno-covered-switch-default
  INCLUDES = -I/home/oystein/build/circt/llvm/llvm/include -I/home/oystein/build/circt/llvm/build/include -I/home/oystein/build/circt/llvm/mlir/include -I/home/oystein/build/circt/llvm/build/tools/mlir/include -I/home/oystein/build/circt/include -I/home/oystein/build/circt/build/include -I/home/oystein/build/circt/build/_deps/slang-src/source/../include -I/home/oystein/build/circt/build/_deps/slang-build/source -isystem /home/oystein/build/circt/build/_deps/slang-src/source/../external -isystem /home/oystein/build/circt/build/_deps/fmt-src/include
  OBJECT_DIR = lib/Conversion/ImportVerilog/CMakeFiles/CIRCTImportVerilog.dir
  OBJECT_FILE_DIR = lib/Conversion/ImportVerilog/CMakeFiles/CIRCTImportVerilog.dir
```
`DEFINES` for the object library object lacks `-DSLANG_BOOST_SINGLE_HEADER -DSLANG_STATIC_DEFINE`. These are not propagated in [llvm_add_library()](https://github.com/llvm/llvm-project/blob/b7d635ed30da49cc32b5b46d00e67ecc3ff9522f/llvm/cmake/modules/AddLLVM.cmake#L565).

I pushed a slightly less confusing hack to https://github.com/llvm/circt/pull/7792/commits/88a8c02fe3b75ad523c008154447c605eba6f31e.

hovind, 2024-11-18:
> > Progress is slow, but I'm working on it every now and then when I find time.
> 
> I'd like to try help if it possible

Fantastic!

Right now, the only failing CI failure is due to https://github.com/llvm/circt/blob/88a8c02fe3b75ad523c008154447c605eba6f31e/.github/workflows/shortIntegrationTests.yml running on ghcr.io/circt/images/circt-integration-test:v15.1, which does not have `clang-17`.

Feel free to help out wherever you would like, but a review of the cumulative changes in https://github.com/llvm/circt/pull/7792/files would be much appreciated.

Flinner, 2025-09-08:
#8835 can be closed?