llvm/circt #7220: CombDataflow op interface should not exist in the HW dialect

prithayan, 2024-06-20:
The CombDataflow op interface is being used only in the FIRRTL dialect, `CheckCombLoops` pass, resolve the downstream dependencies to ensure the interface can be defined in the FIRRTL dialect.
Related to: https://github.com/llvm/circt/pull/7195/files

mortbopet, 2024-06-21:
Drive-by comment: Alternatively, something like [CheckCombLoops](https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Transforms/CheckCombLoops.cpp) seems like a perfectly applicable and generally very useful pass to have at a lower level... so perhaps it isn't the interface that needs to move (may be it needs to move to `comb`), but instead the pass that needs to be made more generic.

sequencer, 2025-06-27:
> but instead the pass that needs to be made more generic.

Agreed, #8610 also illustrated this. If some Op is not handled in `firrtl`, and delegate to `hw`, then `CheckCombLoops` goes wrong.