llvm/circt #7227: [FIRRTL] DPI type check should not check the reference type.

sequencer, 2024-06-23:
Think about this case: 
```firrtl
module A:
  wire a : UInt<64>
  wire b : UInt<256>[256]
  intrinsic(circt_dpi_call<functionName = "some_func", isClocked = 1>, clock, enable, a, b)
module B:
  wire a : UInt<64>
  wire b : UInt<128>[256]
  intrinsic(circt_dpi_call<functionName = "some_func", isClocked = 1>, clock, enable, a, b)
```
`some_func` in these two different module should be lowered to a same function, despite `A.b` and `B.b` has different types:
```c
    extern void some_func(long long in_0, svBitVecVal* out_0);
```
However firtool will complain: 
```
'firrtl.int.dpi.call' op DPI function "some_func" input types don't match
```
related code here:
https://github.com/llvm/circt/blob/90954e2ae8cd7c230755e83754a87af52bfb8a6e/lib/Dialect/FIRRTL/Transforms/LowerDPI.cpp#L146

This is used to dedup DPI-C functions, e.g. DPI call from different AXI agents, but with different datawidth. The payload type in each might be different, but they can still dedup.

uenoku, 2024-06-23:
Good question but unfortunately SV spec doesn't allow that. Verilator will cause a compile error (and VCS emits a warning that "it might incorrectly compile"). If you import DPI functions with a same function name, the input types must be exactly same in System Verlog level.

sequencer, 2024-06-23:
That’s a sad story. I think I have to duplicate that now. Thanks for your answer!

closed 2024-06-23 by sequencer