llvm/circt #7228: [FIRRTL] Constant will be elimated on the DPI Path

sequencer, 2024-06-23:
I'm not sure if this is a bug. If a constant e.g. `channelId` propagating to DPI function, it will be eliminated. I'm currently using a `dontTouch` to work around, but I think this should be resolved in CIRCT.

dtzSiFive, 2024-06-26:
Can you provide an example or maybe elaborate a bit?

What does it mean that they are eliminated? They're fed to DPI calls but the inputs to the DPI function are dropped (?) if the input is a constant, ...?

uenoku, 2024-07-02:
Yeah could you elaborate a bit more?

sequencer, 2024-07-02:
Sorry for the late reply, I'll try to give a reproducible example these two days.

unlsycn, 2024-08-25:
This may be a revalant example:
``` firrtl
FIRRTL version 4.0.0
circuit GCDTestBench :
  layer Verification, bind, "Verification" :
    layer Assert, bind, "Verification/Assert" :
    layer Assume, bind, "Verification/Assume" :
    layer Cover, bind, "Verification/Cover" :
  public module GCDTestBench : 
    input clock : Clock 
    input reset : UInt<1> 

    regreset simulationTime : UInt<64>, clock, reset, UInt<64>(0h0) 
    node _simulationTime_T = add(simulationTime, UInt<1>(0h1)) 
    node _simulationTime_T_1 = tail(_simulationTime_T, 1) 
    connect simulationTime, _simulationTime_T_1 
    node _fooRet_T = eq(simulationTime, UInt<7>(0h64)) 
    node fooRet = intrinsic(circt_dpi_call<functionName = "foo", isClocked = 0> : UInt<8>, _fooRet_T) 
    node _T = neq(fooRet, UInt<1>(0h0)) 
    when _T : 
      layerblock Verification : 
        node _T_1 = eq(reset, UInt<1>(0h0)) 
        when _T_1 : 
          printf(clock, UInt<1>(0h1), "%d:%d}\n", fooRet, simulationTime) : printf 
          stop(clock, UInt<1>(0h1), 0) : stop 

    node _dontTouchRet_T = eq(simulationTime, UInt<7>(0h64)) 
    node dontTouchRet = intrinsic(circt_dpi_call<functionName = "dontTouchMe", isClocked = 0> : UInt<8>, _dontTouchRet_T) 
    node _T_2 = neq(dontTouchRet, UInt<1>(0h0)) 
    when _T_2 : 
      layerblock Verification :  
        node _T_3 = eq(reset, UInt<1>(0h0)) 
        when _T_3 : 
          printf(clock, UInt<1>(0h1), "%d:%d}\n", fooRet, simulationTime) : printf_1 
          stop(clock, UInt<1>(0h1), 0) : stop_1 

    node _barRet_T = eq(simulationTime, UInt<7>(0h64)) 
    node barRet = intrinsic(circt_dpi_call<functionName = "bar", isClocked = 0> : UInt<8>, _barRet_T) 
    node _T_4 = neq(barRet, UInt<1>(0h0)) 
    when _T_4 : 
      node _T_5 = eq(reset, UInt<1>(0h0)) 
      when _T_5 : 
        printf(clock, UInt<1>(0h1), "%d:%d}\n", barRet, simulationTime) : printf_2 
    node _T_6 = eq(simulationTime, UInt<7>(0h64)) 
    node _T_7 = intrinsic(circt_dpi_call<functionName = "baz", isClocked = 0> : UInt<8>, _T_6) 
```
```
[
  {
    "class":"firrtl.transforms.DontTouchAnnotation",
    "target":"~GCDTestBench|GCDTestBench>dontTouchRet"
  }
]
```
The when block that contains `stop` will be elimated while those that don't are not. And dontTouch can keep the signal but when block are still optimized away.

uenoku, 2024-08-25:
Thank you for the example.

> The when block that contains stop will be elimated 

I'm not sure this is eliminated. These blocks are emitted under Verification layer whereas DPI intrinsics are not currently emitted under layer (probably we should emit this under verification layer). Maybe I'm missing something but could you check the issue is related to DPI or Layer?

```
module GCDTestBench_Verification(
  input        _GEN,
               reset,
               clock,
  input [63:0] simulationTime
);

  reg [7:0] _GEN_0;
  always_comb begin
    if (_GEN) begin
      foo(_GEN_0);
    end
    else
      _GEN_0 = 8'bx;
  end // always_comb
  reg [7:0] _GEN_1;
  always_comb begin
    if (_GEN) begin
      dontTouchMe(_GEN_1);
    end
    else
      _GEN_1 = 8'bx;
  end // always_comb
  `ifndef SYNTHESIS
    always @(posedge clock) begin
      automatic logic _GEN_2;
      automatic logic _GEN_3;
      _GEN_2 = (|_GEN_0) & ~reset;
      _GEN_3 = (|_GEN_1) & ~reset;
      if ((`STOP_COND_) & _GEN_2)
        $finish;
      if ((`PRINTF_COND_) & _GEN_2)
        $fwrite(32'h80000002, "%d:%d}\n", _GEN_0, simulationTime);
      if ((`PRINTF_COND_) & _GEN_3)
        $fwrite(32'h80000002, "%d:%d}\n", _GEN_0, simulationTime);
      if ((`STOP_COND_) & _GEN_3)
        $finish;
    end // always @(posedge)
  `endif // not def SYNTHESIS
endmodule
```

unlsycn, 2024-08-25:
> Thank you for the example.
>
>
> I'm not sure this is eliminated. These blocks are emitted under Verification layer whereas DPI intrinsics are not currently emitted under layer (probably we should emit [...]

Thank you, I forgot to enable Verification Layer.

closed 2024-08-25 by sequencer