llvm/circt #7233: [ESI] Add target-attached include directory for cosim targets

mortbopet, 2024-06-25:
Currently, cosim targets rely on on a global `include_directories` call to be able to find include files:
https://github.com/llvm/circt/blob/88eeb265f479ab9cd7800d9859dd623f4d90c891/lib/Dialect/ESI/runtime/cosim/CMakeLists.txt#L56

This makes it difficult for out-of-tree project, which may rely on linking against e.g. `EsiCosimGRPC`, to pick up the includes.

Possible fixes would be to either set that include directory directly on the `EsiCosimGRPC` library, or adding a new intermediate (non-gRPC) library which has the include directory set on it.

teqdruid, 2024-06-28:
#7241 solves this problem.

teqdruid, 2024-07-01:
I think this was taken care of in #7241

closed 2024-07-01 by teqdruid