llvm/circt #7250: [FIRRTL] Move `lib/Firtool` and `lib/Reduce` under `lib/Tools`?

uenoku, 2024-06-27:
`Firtool` living under `lib` looks a bit weird to me. MLIR uses `lib/Tools` as a place to put tool-specific libraries so I think we should  move them to `lib/Tools`.

dtzSiFive, 2024-06-27:
Our dialects depend on CIRCTReduceLib -- totally open to reorg (thank you!!), but not sure it makes sense for this to go into `Tools`.   Looks like upstream MLIR has something kinda similar re:`lib/Reducer`, with the tool-specific portion backing mlir-reduce going into `lib/Tools` as you say.

`lib/Firtool` does seem a bit strange, 💯 , and I think it absolutely makes sense to put under a `Tools` directory.

dtzSiFive, 2024-06-27:
I'm not 100% why our dialects depend on that, where in upstream MLIR they have it split, but haven't looked/thought much yet.

And to complete that thought from earlier a bit more -- probably anything we link into our Dialects doesn't belong in `lib/Tools`, was the thinking.  That's for bits that back the tools 😄 .