llvm/circt #7252: [FIRRTL] OM/InnerSyms/Zero-width

darthscsi, 2024-06-28:
If you have an OMReference to a zero-width wire, when lower-to-hw runs, it drops the wire, but should error out when it needs to drop a wire with a inner symbol.  There is no construct to put the symbol on to maintain the hierpath op for OM.

Even now that HW supports zero-width types, if we used them for firrtl, this problem would still persist as there is no verilog output for a zero-width thing, so there would be no path in the output corresponding to the OMReference.

Fundamentally this was a design bug (non-deletable Reference to a zero-bit thing), but we need to catch it and error out in lower-to-hw rather than making dangling symbol references.

youngar, 2024-06-28:
Duplicate of https://github.com/llvm/circt/issues/5590

darthscsi, 2024-07-01:
closed in https://github.com/llvm/circt/commit/643571890fa0674af1b49aa98413c7babb0862b1

closed 2024-07-01 by darthscsi