llvm/circt #7255: Non-linear commit history.

hailongSun2000, 2024-07-01:
I'm so sorry to break down the commit history. I'm so sorry to cause the unnecessary trouble. :sweat: When I merged this PR https://github.com/llvm/circt/pull/7216, I just rebased the commit history in my local branch, and forgot to push it. I'm so sorry for this mistake. Anyone can help me? Very very thanks! And I'm so sorry for this. I'll be careful later. And take this as a warning.

hailongSun2000, 2024-07-01:
@darthscsi @teqdruid @seldridge @fabianschuiki @uenoku . I'm so sorry to bother you.  Please help me to fix it.

hailongSun2000, 2024-07-01:
The latest commit history based on this(ce121a0a15404062ced9b560c3df53d52f00b1ee)

uenoku, 2024-07-01:
I feel the PR is merged correctly with squash and merge. What is the issue specifically?

teqdruid, 2024-07-01:
Yeah, I don't see the issue.

hailongSun2000, 2024-07-01:
When I merged this PR, I forgot to synchronize the commit history.

hailongSun2000, 2024-07-01:
I checked this comparison (https://github.com/llvm/circt/compare/e70dcc003a0558e94d6b127b9fbbed2dba0cf268..955f484183727c851e2607a6d2ecdb96bfc243ed). There are no changes since this commit(https://github.com/llvm/circt/commit/ce121a0a15404062ced9b560c3df53d52f00b1ee)

uenoku, 2024-07-01:
It’s not a problem unless you don’t force push to a main branch directly. Since you merged a PR through GitHub it was automatically squashed and merge to main branch.

hailongSun2000, 2024-07-01:
Ok, I get it. But I'll be careful and avoid this mistake occurring again.

uenoku, 2024-07-01:
It would be great to rebase a PR to top before merging a PR but please don't worry too much even when you forget that. If CI gets broken unintentionally then we can always revert the commit :+1: Thank you for working hard on improving VerilogImport flow :pray:

closed 2024-07-01 by hailongSun2000