llvm/circt #7308: Perf measurements link (https://circt.org/perf/) broken

mgielda, 2024-07-11:
On the bottom of https://circt.llvm.org/ it says "Nightly performance plots: https://circt.org/perf/" but this does not work. Given that the main page is at https://circt.llvm.org/ I tried https://circt.llvm.org/perf/ but that is a 404.

If the data is available elsewhere, the link could be fixed, or if it is not gathered anywhere, that line could be deleted from the docs.

dtzSiFive, 2024-07-11:
Thanks for reporting these issues!

The data is here: https://github.com/circt/perf FWIW.
I think `https://circt.org/perf` is the correct link (not incorrect anyway) but something else is currently broken.  I'm not entirely sure how this is organized.

Looks like `circt.org` no longer "working" (configured nameservers don't have information for it), which seems like part of the problem...

ecstrema, 2025-01-20:
This is still broken.

Maybe just remove the perf link and point directly to the Girhub repo?