llvm/circt #7384: [FIRRTL] in `replSeqMem` stands for "replace sequential memories" not "replicate sequential memories"

youngar, 2024-07-24:
We have some flags named `shouldReplicateSequentialMemories`, but AFAICT this is incorrectly named and should be called `shouldReplaceSequentialMemories`.  See the search result: https://github.com/search?q=repo%3Allvm%2Fcirct%20shouldReplicateSequentialMemories&type=code

jpien13, 2024-08-02:
Hey, I was referred to this issue by colleagues of Megan Wach. I am a rising senior at college studying computer engineering. This would be my first issue within the CIRCT project. I'd be happy to try and tackle this one.

jpien13, 2024-08-04:
Hey I made the changes and it looks like I need permissions to push my branch:

remote: Permission to llvm/circt.git denied to jpien13.
fatal: unable to access 'https://github.com/llvm/circt.git/': The requested URL returned error: 403

youngar, 2024-08-04:
Hi @jpien13, thanks for working on this!  To get around the issue, you should fork circuit, push your branch to your fork, and then open a cross-fork PR from your fork to `llvm/circt`. You might find [this](https://llvm.org/docs/GitHub.html) guide helpful. Once you have a some commits under your belt you can [ask to be added to the llvm project](https://llvm.org/docs/DeveloperPolicy.html#obtaining-commit-access).

jpien13, 2024-08-05:
@youngar just made the PR! Not sure what the review process is but let me know if you have any feedback :) thanks!