llvm/circt #7399: [ExportVerilog] Comb ExtractOp assertion "should be handled by isExpressionUnableToInline" failed with lowering option disallowLocalVariables

7FM, 2024-07-29:
To reproduce the failing assertion, use the following mlir code and run `circt-opt -hw-legalize-modules  -export-verilog <test.mlir>`:
```mlir
module @export_verilog_crash attributes {circt.loweringOptions = "disallowLocalVariables"} {

  hw.module @bitextract_crash(in %clk : i1, in %sel : i3, in %dummy : i32, out MyOutput : i17) {
    %res_reg = sv.reg sym @res_reg : !hw.inout<i17> 

    sv.alwaysff(posedge %clk) {
      %c-1209970334_i32 = hw.constant -1209970334 : i32
      %c-1028445891_i32 = hw.constant -1028445891 : i32

      %casez_tmp = sv.reg : !hw.inout<i32> 
      sv.case casez %sel : i3
      case b000: {
        sv.bpassign %casez_tmp, %c-1209970334_i32 : i32
      }
      default: {
        sv.bpassign %casez_tmp, %c-1028445891_i32 : i32
      }
      %17 = sv.read_inout %casez_tmp : !hw.inout<i32>
      %29 = comb.xor %dummy, %17 : i32
      %32 = comb.extract %29 from 3 : (i32) -> i17
      sv.passign %res_reg, %32 : i17
    }

    %res_reg_data = sv.read_inout %res_reg : !hw.inout<i17>
    hw.output %res_reg_data : i17
  }
}
```
Note that the `-hw-legalize-modules` is optional to reproduce the crash.
`circt.loweringOptions = "disallowLocalVariables"` is, however, required.

Here is the full stacktrace:
```
circt-opt: circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:2754: SubExprInfo (anonymous namespace)::ExprEmitter::visitComb(ExtractOp): Assertion `(x.precedence == Symbol || (x.precedence == Selection && isOkToBitSelectFrom(op.getInput()))) && "should be handled by isExpressionUnableToInline"' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
 #0 0x000055589ad931bc llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) circt/llvm/llvm/lib/Support/Unix/Signals.inc:723:13
 #1 0x000055589ad90df0 llvm::sys::RunSignalHandlers() circt/llvm/llvm/lib/Support/Signals.cpp:106:18
 #2 0x000055589ad93935 SignalHandler(int) circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #3 0x00007f64ccc52f30 __restore_rt (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x3ff30)
 #4 0x00007f64ccca2efc __pthread_kill_implementation (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x8fefc)
 #5 0x00007f64ccc52e86 gsignal (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x3fe86)
 #6 0x00007f64ccc3b935 abort (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x28935)
 #7 0x00007f64ccc3b859 _nl_load_domain.cold (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x28859)
 #8 0x00007f64ccc4b6e6 (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x386e6)
 #9 0x000055589b1dc6f7 decltype(auto) llvm::cast<mlir::detail::TypedValue<mlir::IntegerType>, mlir::Value>(mlir::Value const&) circt/llvm/llvm/include/llvm/Support/Casting.h:566:3
#10 0x000055589b1dc6f7 circt::comb::MuxOp::getCond() circt/build/include/circt/Dialect/Comb/Comb.h.inc:2455:12
#11 0x000055589b1dc6f7 (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()::operator()() const circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:3222:22
#12 0x000055589b1dc6f7 void std::__invoke_impl<void, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>(std::__invoke_other, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/bits/invoke.h:61:14
#13 0x000055589b1dc6f7 std::__invoke_result<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>::type std::__invoke<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>((anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/bits/invoke.h:96:14
#14 0x000055589b1dc6f7 std::invoke_result<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>::type std::invoke<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>((anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/functional:113:14
#15 0x000055589b1dc6f7 auto circt::pretty::TokenStream<circt::pretty::BufferingPP>::scopedBox<circt::pretty::PP, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()>(circt::pretty::PP&&, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const::'lambda'()&&, circt::pretty::Token) circt/include/circt/Support/PrettyPrinterHelpers.h:370:12
#16 0x000055589b1dc6f7 (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0::operator()() const circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:3221:8
#17 0x000055589b1dc6f7 (anonymous namespace)::SubExprInfo std::__invoke_impl<(anonymous namespace)::SubExprInfo, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>(std::__invoke_other, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/bits/invoke.h:61:14
#18 0x000055589b1dc6f7 std::__invoke_result<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>::type std::__invoke<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>((anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/bits/invoke.h:96:14
#19 0x000055589b1dc6f7 std::invoke_result<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>::type std::invoke<(anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>((anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0&&) /nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/include/c++/13.2.0/functional:113:14
#20 0x000055589b1dc6f7 auto circt::pretty::TokenStream<circt::pretty::BufferingPP>::scopedBox<circt::pretty::PP, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0>(circt::pretty::PP&&, (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp)::$_0&&, circt::pretty::Token) circt/include/circt/Support/PrettyPrinterHelpers.h:370:12
#21 0x000055589b1dc6f7 (anonymous namespace)::ExprEmitter::visitComb(circt::comb::MuxOp) circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:3220:13
#22 0x000055589b1dc6f7 (anonymous namespace)::SubExprInfo circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::comb::MuxOp>(auto) const circt/include/circt/Dialect/Comb/CombVisitors.h:43:32
#23 0x000055589b1dc6f7 llvm::TypeSwitch<mlir::Operation*, (anonymous namespace)::SubExprInfo>& llvm::TypeSwitch<mlir::Operation*, (anonymous namespace)::SubExprInfo>::Case<circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)&) circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:102:22
#24 0x000055589b1dc6f7 circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*) circt/include/circt/Dialect/Comb/CombVisitors.h:30:19
#25 0x000055589b1dc6f7 (anonymous namespace)::ExprEmitter::emitSubExpr(mlir::Value, (anonymous namespace)::VerilogPrecedence, (anonymous namespace)::ExprEmitter::SubExprSignRequirement, bool, bool) circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:2633:18
#26 0x000055589b1d9b70 circt::pretty::TokenBuilder<circt::pretty::BufferingPP>::addToken(circt::pretty::Token) circt/include/circt/Support/PrettyPrinterHelpers.h:93:28
#27 0x000055589b1d9b70 circt::pretty::TokenStream<circt::pretty::BufferingPP>::operator<<(circt::pretty::Token) circt/include/circt/Support/PrettyPrinterHelpers.h:324:11
[... cut ...]
#111 0x000055589ce25c04 void* llvm::thread::ThreadProxy<std::tuple<llvm::StdThreadPool::grow(int)::$_0>>(void*) circt/llvm/llvm/include/llvm/Support/thread.h:55:5
#112 0x00007f64ccca1272 start_thread (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x8e272)
#113 0x00007f64ccd1cdcc __GI___clone3 (/nix/store/ddwyrxif62r8n6xclvskjyy6szdhvj60-glibc-2.39-5/lib/libc.so.6+0x109dcc)
Aborted (core dumped)
```
The corresponding assertion:
https://github.com/llvm/circt/blob/01d98e8fa0770ad978c4f8fafc56f17a889a75be/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L2752-L2754

uenoku, 2024-07-30:
Thank you for a great reproducer! Should be fixed by #7404.

7FM, 2024-07-30:
> Thank you for a great reproducer! Should be fixed by #7404.

Awesome, thx for the fast fix!