llvm/circt #7406: [hwarith] Shouldn't ICmpOp result be `i1`?

teqdruid, 2024-07-30:
The result of a `hwarith.icmp` is always `ui1`. Shouldn't it be `i1`? Logically, the result is a boolean which doesn't have signedness semantics. What was the original rationale for this (since it seems like a very intentional decision)?

teqdruid, 2024-07-30:
@mortbopet @jopperm

jopperm, 2024-07-30:
Hmm, I don't remember whether we discussed that specifically (CC: @7FM). I think the idea was that `hwarith`'s operations would work on MLIR's signed/unsigned types, with the `cast` operation being the defined transition point to/from the signless world, so `ui1` was the natural choice. But yeah, you can't do much with the result in the dialect other than casting to `i1`...

teqdruid, 2024-07-31:
Ok. Ima change it then.

commit d6a0d35b (2024-07-31): [HWArith] Make `hwarith.icmp` result an `i1`

As discussed in #7406. Resolves #7406.
PR #7413 (2024-08-09): [HWArith] Make `hwarith.icmp` result an `i1`
As discussed in #7406. Resolves #7406.
commit 562036cd (2024-08-09): [HWArith] Make `hwarith.icmp` result an `i1` (#7413)

As discussed in #7406. Resolves #7406.
closed 2024-08-09 by teqdruid