llvm/circt #7410: [SV] custom op 'sv.ifdef' invalid kind of attribute specified.

hailongSun2000, 2024-07-31:
When I ran arc-tests, I met an error as follow:
``` c
make -C boom run
make: Entering directory '/home/phoenix/Downloads/arc-tests/arc-tests/boom'
gzip -dc boom-small.fir.gz > build/small/boom.fir
firtool --ir-hw build/small/boom.fir -o build/small/boom.mlir
arcilator build/small/boom.mlir --state-file=build/small/boom.json -o build/small/boom-arc.ll --mlir-timing --print-debug-info --mlir-pass-statistics --observe-wires=0 --observe-ports=0 --observe-named-values=0 --observe-registers=0 --observe-memories=0
build/small/boom.mlir:3:13: error: custom op 'sv.ifdef' invalid kind of attribute specified
  sv.ifdef  "RANDOMIZE" {
            ^
build/small/boom.mlir:3:25: error: custom op 'sv.ifdef' failed to parse MacroIdentAttr parameter 'ident' which is to be a `::mlir::FlatSymbolRefAttr`
  sv.ifdef  "RANDOMIZE" {
                        ^
===-------------------------------------------------------------------------===
                         ... Execution time report ...
===-------------------------------------------------------------------------===
  Total Execution Time: 0.0426 seconds

  ----Wall Time----  ----Name----
    0.0418 ( 98.0%)  Parse MLIR input
    0.0008 (  2.0%)  Rest
    0.0426 (100.0%)  Total
make: *** [Makefile:42: build/small/boom.json] Error 1
make: Leaving directory '/home/phoenix/Downloads/arc-tests/arc-tests/boom'
```

uenoku, 2024-07-31:
I think you are using different version of firtool and arcilator (probably firtool is super old).

hailongSun2000, 2024-07-31:
It makes sense! Let me update and try again.

hailongSun2000, 2024-07-31:
I update firtool to the latest, but there is an another error regarding the FIRRTL version
```c
make: Entering directory '/home/phoenix/Downloads/arc-tests/arc-tests/boom'
gzip -dc boom-small.fir.gz > build/small/boom.fir
firtool --ir-hw build/small/boom.fir -o build/small/boom.mlir
build/small/boom.fir:1:16: error: FIRRTL version must be >=2.0.0
FIRRTL version 1.1.0
```

uenoku, 2024-07-31:
I think the FIR file is generated a long time ago so probably the latest firtool dropped support for that. We will need to re-generate the test suites.

hailongSun2000, 2024-07-31:
Thanks so much for your help. I successfully rebuilt the rocket, but the generated rocket.fir lost FIRRTL :sweat_smile: , like
```
build/small-v1.6/rocket.fir:1:1: error: expected 'FIRRTL'
circuit RocketSystem :
^
make: *** [Makefile:36: build/small-v1.6/rocket.mlir] Error 1
```

uenoku, 2024-07-31:
I think Chisel also needs to be bumped to recent version. It looks rocket-chip repo uses chisel 5 but I believe we need to bump 6 or 7 (don't remember the exact version).

hailongSun2000, 2024-07-31:
You're right! Thanks so much :smiley:!

closed 2024-08-01 by hailongSun2000