llvm/circt #7414: [ExportVerilog][Verif] 'verif.assert' op emission to Verilog not supported

teqdruid, 2024-07-31:
When I run the `check-circt` tests locally, I the below on the `Conversion/ExportVerilog/prepare-for-emission.mlir` test. For some reason, this doesn't fail in CI.

```
--
Command Output (stderr):
--
RUN: at line 1: /home/jodemme/_circt_build/bin/circt-opt /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir --pass-pipeline='builtin.module(any(prepare-for-emission))' --split-input-file -verify-diagnostics | /home/jodemme/_circt_build/bin/FileCheck /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir
+ /home/jodemme/_circt_build/bin/circt-opt /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir '--pass-pipeline=builtin.module(any(prepare-for-emission))' --split-input-file -verify-diagnostics
+ /home/jodemme/_circt_build/bin/FileCheck /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir
RUN: at line 2: /home/jodemme/_circt_build/bin/circt-opt /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir -export-verilog -split-input-file
+ /home/jodemme/_circt_build/bin/circt-opt /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir -export-verilog -split-input-file
within split at /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir:149 offset :16:5: error: 'verif.assert' op emission to Verilog not supported
    verif.assert %k0 : !ltl.property
    ^
within split at /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir:149 offset :16:5: note: see current operation: "verif.assert"(%4) : (!ltl.property) -> ()
within split at /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir:256 offset :7:3: error: 'verif.assert' op emission to Verilog not supported
  verif.assert %2 : !ltl.sequence
  ^
within split at /home/jodemme/circt/test/Conversion/ExportVerilog/prepare-for-emission.mlir:256 offset :7:3: note: see current operation: "verif.assert"(%2) : (!ltl.sequence) -> ()
```

I'm running clang-10 on Ubuntu 20.04.

Extracted the split which fails.

```
hw.module @Issue5613(in %a: i1, in %b: i1) {
  verif.assert %2 : !ltl.sequence
  %0 = ltl.implication %2, %1 : !ltl.sequence, !ltl.property
  %1 = ltl.or %b, %3 : i1, !ltl.property
  %2 = ltl.and %b, %4 : i1, !ltl.sequence
  %3 = ltl.not %b : i1
  %4 = ltl.delay %a, 42 : i1
  hw.output
}
```

```
$ circt-opt issue.mlir -export-verilog
// Generated by CIRCT unknown git version
module Issue5613(       // issue.mlir:2:1
  input a,      // issue.mlir:2:25
        b       // issue.mlir:2:36
);issue.mlir:3:3: error: 'verif.assert' op emission to Verilog not supported
  verif.assert %2 : !ltl.sequence
  ^
issue.mlir:3:3: note: see current operation: "verif.assert"(%2) : (!ltl.sequence) -> ()

  unknown MLIR operation verif.assert
endmodule
```

teqdruid, 2024-07-31:
When I add the `--lower-verif-to-sv` pass, it works fine. I checked and I cannot find ExportVerilog support for any verif ops. So AFAICT this test should fail, so the question may be why does it not fail in CI?

fzi-hielscher, 2024-07-31:
Looks like `-split-input-file` is dropping the failure exit code. Testing it locally (MacOS):

```
> ./bin/circt-opt ../test/Conversion/ExportVerilog/prepare-for-emission.mlir -export-verilog &> /dev/null
> echo $?
1
> ./bin/circt-opt ../test/Conversion/ExportVerilog/prepare-for-emission.mlir -export-verilog -split-input-file &> /dev/null
> echo $?
0
```

That's not how it should behave, is it? 🫢

teqdruid, 2024-07-31:
Dunno. I never use `-split-input-file` unless I'm using `-verify-diagnostics` also.

fzi-hielscher, 2024-07-31:
Never mind, it had nothing to do with `-split-input-file`. It fails without splitting because of the different top-level operations. And it does not fail with splitting because `StmtEmitter::emitStatement(Operation *op)` does emit an error but does not signal failure (on purpose I guess?!?). 😵‍💫

teqdruid, 2024-07-31:
That's intentional, but IMO not desirable (anymore).

dtzSiFive, 2024-08-01:
Oh I didn't know it was intentional, but yes I don't think it's desirable.

cc #4770 which is another but probably there are more... these should fail the pass (and the tool they're part of, accordingly).

teqdruid, 2024-08-09:
Mitigated by [46e1297](https://github.com/llvm/circt/commit/46e129756eb6b2eb4e9c949fbf3742180663909d)

closed 2024-08-09 by teqdruid