llvm/circt #7415: [FIRRTL][Dedup] Incorrect deduplication of same operations in different blocks

dtzSiFive, 2024-07-31:
Consider:
```firrtl
firrtl.circuit "Whens" {
  firrtl.module private @A(in %x: !firrtl.uint<1>) {
    firrtl.when %x : !firrtl.uint<1> {
      %0 = firrtl.constant 5 : !firrtl.uint<32>
      %n = firrtl.node %0 : !firrtl.uint<32>
    }
  }
  firrtl.module private @B(in %x: !firrtl.uint<1>) {
    firrtl.when %x : !firrtl.uint<1> {
    } else {
      %0 = firrtl.constant 5 : !firrtl.uint<32>
      %n = firrtl.node %0 : !firrtl.uint<32>
    }
  }
  firrtl.module @Whens(in %x: !firrtl.uint<1>) {
    %a_x = firrtl.instance a @A(in x: !firrtl.uint<1>)
    firrtl.matchingconnect %a_x, %x : !firrtl.uint<1>
    %b_x = firrtl.instance b @B(in x: !firrtl.uint<1>)
    firrtl.matchingconnect %b_x, %x : !firrtl.uint<1>
  }
}
```

Note that the first `firrtl.when` has two operations in the "true" block, the second has the same two in the "false" lblock.

`circt-opt --firrtl-dedup` on the above yields:

```firrtl
firrtl.circuit "Whens" {
  firrtl.module private @A(in %x: !firrtl.uint<1>) {
    firrtl.when %x : !firrtl.uint<1> {
      %c5_ui32 = firrtl.constant 5 : !firrtl.uint<32>
      %n = firrtl.node %c5_ui32 : !firrtl.uint<32>
    }
  }
  firrtl.module @Whens(in %x: !firrtl.uint<1>) {
    %a_x = firrtl.instance a @A(in x: !firrtl.uint<1>)
    firrtl.matchingconnect %a_x, %x : !firrtl.uint<1>
    %b_x = firrtl.instance b @A(in x: !firrtl.uint<1>)
    firrtl.matchingconnect %b_x, %x : !firrtl.uint<1>
  }
}
```

dtzSiFive, 2024-08-01:
FIRRTL example:

```firrtl
FIRRTL version 4.0.0

circuit Block: %[[
{ "class": "firrtl.passes.InlineAnnotation",
  "target": "~Block|A" },
{ "class": "firrtl.passes.InlineAnnotation",
  "target": "~Block|B" }
]]
  module A:
    input x : UInt<1>
    output y : UInt<1>

    connect y, UInt<1>(0)

    when x :
      node n = x
      connect y, UInt<1>(1)
    else :
      skip

  module B:
    input x : UInt<1>
    output y : UInt<1>

    connect y, UInt<1>(0)

    when x :
      node n = x
    else :
      connect y, UInt<1>(1)
      skip

  public module Block:
    input x : UInt<1>
    output y : UInt<1>

    inst a of A
    connect a.x, x
    inst b of B
    connect b.x, x

    connect y, and(a.y, b.y)
```

Presently firtool compiles this to:
```
// Generated by CIRCT firtool-1.79.0g20240729_15417e8
module Block(
  input  x,
  output y
);

  assign y = x;
endmodule
```

But it should be:

```
// Generated by CIRCT firtool-1.79.0-22-ge99f315ca
module Block(
  input  x,
  output y
);

  assign y = 1'h0;
endmodule
```

More creative examples can invert assertions and so on.

seldridge, 2024-08-01:
For the curious, you can exercise this by constructing a pass pipeline that runs `ExpandWhens` before `Dedup`:

```
firtool 7415.fir -parse-only | circt-opt -pass-pipeline='builtin.module(firrtl.circuit(firrtl.module(firrtl-expand-whens)))' | circt-opt -firrtl-dedup | firtool -format=mlir
```

closed 2024-08-01 by dtzSiFive

dtzSiFive, 2024-08-01:
Thanks for the snippet! Yep manually running expand-when's first will help produce the correct output on this input using tools containing the bugged dedup.

PR #7420 (2024-08-01): [FIRRTL][Dedup] Rework hashing for perf and bug fixes.
Primary change is to only generate and populate mappings for sources of values, and not each value themselves.

Values are identified using these base numberings plus an appropriate offset.

The main benefit of this is to greatly reduce the number of entries in the `indices` map.
When handling operations with many block arguments (module-like's with many ports) or with many results (instances of those module-like's) this greatly reduces the pressure on the `indices` map.  For these designs, dedup now runs dramatically faster and uses signi [... cut]
commit 20cb546d (2024-08-01): [FIRRTL][Dedup] Rework hashing for perf and bug fixes. (#7420)

Primary change is to only generate and populate mappings for
sources of values, and not each value themselves.

Values are identified using these base numberings plus an appropriate
offset.

The main benefit of this is to greatly reduce the number of entries in the
`indices` map.
When handling operations with many block arguments (module-like's with many
ports) or with many results (instances of those module-like's) this greatly
reduces the pressure on the `indices` map.  For these designs, dedup now runs
dramatically faster and u [... cut]