llvm/circt #7431: [ImportVerilog] Create variables for function arguments

fabianschuiki, 2024-08-03:
The following SV currently crashes because `x` is lowered to an `i32`, which is unassignable, instead of a `ref<i32>`, which would be assignable:
```
function void foo(int x);
  x = 0;
endfunction
```
To fix this, create local variables to carry the function argument values.

Max-astro, 2024-11-07:
Hi fabianschuiki,

I recently encountered the same crash you reported. I’m interested in working on this fix and learning more about the CIRCT project, as I’m still a beginner here.

Where you suggested that "create local variables that shadow the function arguments." might be a fix. I reviewed the crash stack, but I'm not fully clear on which function I need to create local variables for function arguments.

Could you explain a bit more about what "shadowing" function arguments involves in this context, or perhaps share an example of what you had in mind? Any additional insights would be greatly appreciated as I work toward understanding and implementing a solution.

Thank you for your help and guidance, and I look forward to hopefully collaborating on this!

fabianschuiki, 2024-11-07:
Hi @Max-astro, thanks a lot for taking a look at this!

The crash happens because `x` is converted to a function argument, which is just a simple value. In order to be able to assign to it, `x` would have to be converted to a variable instead. This is all handled in the `Context::convertFunction` function in ImportVerilog, in the `lib/Conversion/ImportVerilog/Structure.cpp` file.

In a nutshell, the current implementation is trying to convert the SV
```systemverilog
function void foo(int x);
  x = 0;
endfunction
```
into the MLIR
```mlir
func.func private @foo(%arg0: !moore.i32) {
  %0 = moore.constant 0 : i32
  moore.assign %arg0, %0 : i32
  return
}
```
But this crashes since `moore.assign` requires the destination of the assignment to be a `!moore.ref<i32>`, so something like a `moore.variable`, instead of a simple `i32` like the function argument.

To fix this, the conversion should produce something like the following instead:
```mlir
func.func private @foo(%arg0: !moore.i32) {
  %x = moore.variable %arg0 : <i32>  // <--- a variable that is initialized to the func arg value
  %0 = moore.constant 0 : i32
  moore.assign %x, %0 : i32  // <--- assignment to the variable instead of the func arg
  return
}
```

In `convertFunction`, when the block arguments for the function body are created (the `%arg0` and so on), you should be able to look at the type of the argument and if it is not a `!moore.ref<...>` type, you should be able to create a `VariableOp` that takes the block argument as initial value, and gives you back a `!moore.ref<...>` that you can store in the `valueSymbols` map. This means that when a user types `x` somewhere else in the code, they will get the `VariableOp` back, which they _can_ assign to, instead of the raw function argument. And if the user never assigns to the variable, the canonicalizers will get rid of it automatically again later 😃

Max-astro, 2024-11-12:
Hi @fabianschuiki, thank you so much for your kind reply, and I'm sorry I took so long to fix it.

Here's my fix:
``` diff
diff --git a/lib/Conversion/ImportVerilog/Structure.cpp b/lib/Conversion/ImportVerilog/Structure.cpp
--- a/lib/Conversion/ImportVerilog/Structure.cpp
+++ b/lib/Conversion/ImportVerilog/Structure.cpp
@@ -900,7 +900,22 @@ Context::convertFunction(const slang::ast::SubroutineSymbol &subroutine) {
                  lowering->op.getFunctionType().getInputs())) {
     auto loc = convertLocation(astArg->location);
     auto blockArg = block.addArgument(type, loc);
-    valueSymbols.insert(astArg, blockArg);
+
+    // valueSymbols.insert(astArg, blockArg);
+    if (isa<moore::RefType>(type)) {
+      valueSymbols.insert(astArg, blockArg);
+    } else {
+      // Convert the body of the function.
+      OpBuilder::InsertionGuard g(builder);
+      builder.setInsertionPointToEnd(&block);
+
+      Value shadowArg = builder.create<moore::VariableOp>(
+          loc, moore::RefType::get(cast<moore::UnpackedType>(type)),
+          StringAttr{}, blockArg);
+      // shadowArg.dump();
+      // shadowArg.getParentBlock()->dump();
+      valueSymbols.insert(astArg, shadowArg);
+    }
   }
```

After that, the output MLIR is almost the same as you mentioned.
``` mlir
func.func private @foo(%arg0: !moore.i32) {
    %0 = moore.variable %arg0 : <i32>
    %1 = moore.constant 0 : i32
    moore.blocking_assign %0, %1 : i32
    return
}
```
However, it’s also causing some test failures, there are some errors when I run "check-circt"
For example in this test case:
`circt-translate --import-verilog test/Conversion/ImportVerilog/basic.sv`

Golden outputs:
```
func.func private @dummyE(%arg0: !moore.i1) -> !moore.i1 {
    return %arg0 : !moore.i1
}
```

New outputs:
```
func.func private @dummyE(%arg0: !moore.i1) -> !moore.i1 {
    %4 = moore.variable %arg0 : <i1>
    %5 = moore.read %4 : <i1>
    return %5 : !moore.i1
}
```

It appears that the canonicalizers did not automatically eliminate these redundant statements.
Could you provide any insight on the test failure? Is there something wrong with my implementation or am I missing something?

Thank you once again for your patience and guidance! 😃

fabianschuiki, 2024-11-12:
Really cool! That output looks great :sunglasses:. I'm pretty sure there are a few test cases which just assumed that the function arguments would be used directly, instead of going through a variable. That's a bug in the tests though -- feel free to adjust the tests to now work with `<arg> = moore.read <shadow-var>` instead. If you feel like it, you could also use the name of the function argument as the name of the shadow variable, which might make it easier to write tests and which could make the IR easier to understand. Modules do a similar thing already :smiley:

commit 7fa86324 (2024-11-21): [ImportVerilog]: Create variables for function arguments (#7829)

Adds shadow variables for SV function arguments.
Fixes: #7431

Adjusts two test cases test/Conversion/ImportVerilog/basic.sv and test/Conversion/ImportVerilog/builtins.sv

Adds two test functions in test/Conversion/ImportVerilog/builtins.sv, which will crash in the former circt-verilog tool.

---------
closed 2024-11-21 by hailongSun2000