llvm/circt #7467: [FIRRTL][Layers] ref.send is rejected but is always passive capture

dtzSiFive, 2024-08-08:
Consider:

```firrtl
FIRRTL version 4.0.0

circuit LayersProbe:
  layer A, bind:

  public module LayersProbe:
    output o : { flip x : UInt<1> }
    output q : Probe<{ x : UInt<1>}, A>

    layerblock A :
      define q = probe(o)
```

firtool produces:

```
layers-probe-test.fir:10:5: error: 'firrtl.layerblock' op captures an operand which is not a passive type
    layerblock A :
    ^
layers-probe-test.fir:10:5: note: see current operation:
"firrtl.layerblock"() ({
  %0 = "firrtl.ref.send"(%arg0) : (!firrtl.bundle<x flip: uint<1>>) -> !firrtl.probe<bundle<x: uint<1>>>
  %1 = "firrtl.ref.cast"(%0) : (!firrtl.probe<bundle<x: uint<1>>>) -> !firrtl.probe<bundle<x: uint<1>>, @A>
  "firrtl.ref.define"(%arg1, %1) : (!firrtl.probe<bundle<x: uint<1>>, @A>, !firrtl.probe<bundle<x: uint<1>>, @A>) -> ()
}) {layerName = @A} : () -> ()
layers-probe-test.fir:7:12: note: operand is defined here
    output o : { flip x : UInt<1> }
           ^
layers-probe-test.fir:11:18: note: operand is used here
      define q = probe(o)
                 ^
```

dtzSiFive, 2024-08-08:
Looks like 7462 fixes this, although LowerLayers has a bug regarding non-passive capture via ref.send .

seldridge, 2024-08-09:
Fixed via #7462.

closed 2024-08-09 by seldridge