llvm/circt #7472: circt-verilog: failed to legalize operation 'moore.variable'

imphil, 2024-08-08:
I'm playing around with the SV frontend, and I have yet to get it to work. What am I doing wrong here? I tried some other variants of the Verilog (without ports, with reg instead of logic, etc.), but the error remains roughly the same.

```sv
// dff.sv

`timescale 1us/1ns

module dff (
    output logic q,
    input logic clk, d
);

always @(posedge clk) begin
    q <= d;
end

endmodule
```

```shell-session
$ ../circt/build/bin/circt-verilog dff.sv
dff.sv:6:18: error: failed to legalize operation 'moore.variable'
    output logic q,
                 ^
dff.sv:6:18: note: see current operation: %0 = "moore.variable"() <{name = "q"}> : () -> !moore.ref<l1>
```

FYI @fabianschuiki

fabianschuiki, 2024-08-08:
Hey @imphil, thanks for giving the SV frontend a try and reporting this :+1:.

`circt-verilog` has a few lowering modes you can select from, which allow you to run the lowering pipeline more or less far. The safest and shortest one is `circt-verilog --parse-only`, which will spit out a whole bunch of MLIR containing `moore.*` operations. There is also `--ir-moore`, which will additionally run a bunch of optimization passes, and `--ir-hw`, which is the default and tries to lower the Verilog input all the way down to CIRCT's core IR dialects.

The optimizations at the Moore dialect level (`--ir-moore`) are somewhat shaky, and there are quite a few things that the Moore-to-Core lowering (`--ir-hw`) doesn't fully handle yet. In your example, the lowering to the core dialects gives up when it sees a variable that isn't just trivially assigned :cry: Sorry about that!

One of the big next milestones it to get the Moore-to-Core conversion to properly convert from the high-level Moore dialect to the low-level LLHD dialect, which includes handling variables and blocking/non-blocking assignments to them as in your example.

imphil, 2024-08-08:
Ah, I see. Thanks for the explanation! Looks like I was simply to early to try this out then. Do you have a rough idea on the timeline/work required to get to this point where the lowering to the core dialect works (for simple examples)?

fabianschuiki, 2024-08-08:
Getting simple examples lowered shouldn't take too long, maybe a couple of days of work. There's a ton of cool optimizations and analyses to be built down the road, such as detecting registers already at the Moore dialect level, etc.. But even without that it should always be possible to just lower always, initial, and final procedures directly to LLHD, and then you'd be able to at least simulate them :smiley:

closed 2024-08-19 by imphil