llvm/circt #7542: [FIRTOOL] Verification layers are not included in filelist.f

FanShupei, 2024-08-22:
Background: We are bumping our firtool version. See https://github.com/chipsalliance/t1/pull/734

# Description

After the bumping, our workflow breaks as if all `$stop` statements are discarded. It seems recently how firtool generates verilog code has drastically changed.

We observe that `layers_TestBench_Verification.sv` and such are NOT included in `filelist.f`. These files contains `bind` statements to actually bind verification modules, thus it effectively throw out all code in verification layers. This is the root cause of the issue.

Any ideas on how to recover the previous behavior?

firtool version after bumping: 1.81.1

# Addtional Context

<details>

<summary>Content of `filelist.f`</summary>

```
extern_modules.sv
LoadUnit.sv
ram_2x32.sv
Queue2_UInt32.sv
Queue2_UInt32_8.sv
ram_17x32.sv
Queue17_UInt32.sv
Verification/Assert/StoreUnit_Verification_Assert.sv
StoreUnit.sv
ram_4x72.sv
Queue4_SimpleAccessStage1.sv
ram_4x32.sv
Queue4_UInt32.sv
Verification/Assert/SimpleAccessUnit_Verification_Assert.sv
SimpleAccessUnit.sv
ram_96x54.sv
Queue96_LSUWriteQueueBundle.sv
ram_16x8.sv
Queue16_UInt8.sv
ram_2x325.sv
Queue2_MemWrite.sv
ram_2x78.sv
Queue2_SimpleMemWrite.sv
LSU.sv
VectorDecoder.sv
VectorAdder32.sv
ReduceAdder.sv
LaneBitLogic.sv
LaneLogic.sv
AddRawFN.sv
RoundAnyRawFNToRecFN_ie8_is26_oe8_os24.sv
RoundRawFNToRecFN_e8_s24.sv
AddRecFN.sv
FloatAdder.sv
CompareRecFN.sv
FloatCompare.sv
ChainingCheck.sv
WriteCheck.sv
Verification/Assert/VRF_Verification_Assert.sv
rfVec_rf_sram_64x36.sv
VRF.sv
Queue1_VRFWriteRequest.sv
ram_4x46.sv
Queue4_VRFWriteRequest.sv
MaskedWrite.sv
ram_5x3.sv
Queue5_UInt3.sv
SlotTokenManager.sv
LaneStage0.sv
ram_4x17.sv
Queue4_VRFReadQueueEntry.sv
ram_14x158.sv
Queue14_LaneStage1Enqueue.sv
ReadStageRRArbiter.sv
Verification/Assert/VrfReadPipe_Verification_Assert.sv
VrfReadPipe.sv
ReadStageRRArbiter_1.sv
Verification/Assert/VrfReadPipe_1_Verification_Assert.sv
VrfReadPipe_1.sv
Queue1_UInt64.sv
CrossReadUnit.sv
ram_14x5.sv
Queue14_UInt5.sv
Verification/Assert/LaneStage1_Verification_Assert.sv
LaneStage1.sv
ram_4x105.sv
Queue4_LaneExecuteStage.sv
LaneStage2.sv
ram_6x104.sv
Queue6_ExecutionBridgeRecordQueue.sv
ram_4x104.sv
Queue4_LaneExecuteResponse.sv
Verification/Assert/LaneExecutionBridge_Verification_Assert.sv
LaneExecutionBridge.sv
Queue4_VRFWriteRequest_2.sv
ram_4x1.sv
Queue4_UInt1.sv
LaneStage3.sv
Queue4_ReadBusData.sv
LaneStage0_1.sv
ram_14x157.sv
Queue14_LaneStage1Enqueue_1.sv
LaneStage1_1.sv
Queue4_LaneExecuteStage_1.sv
LaneStage2_1.sv
ram_6x103.sv
Queue6_ExecutionBridgeRecordQueue_1.sv
ram_4x38.sv
Queue4_LaneExecuteResponse_1.sv
Verification/Assert/LaneExecutionBridge_1_Verification_Assert.sv
LaneExecutionBridge_1.sv
LaneStage3_1.sv
Verification/Assert/LaneExecutionBridge_2_Verification_Assert.sv
LaneExecutionBridge_2.sv
Verification/Assert/LaneExecutionBridge_3_Verification_Assert.sv
LaneExecutionBridge_3.sv
MaskedLogic.sv
LaneAdder.sv
LaneShifter.sv
Abs32.sv
CSACompressor4_2.sv
CSA42.sv
CSA42_1.sv
CSA42_2.sv
CSA42_3.sv
CSA42_4.sv
Multiplier16.sv
CSA42_5.sv
CSA42_6.sv
VectorAdder64.sv
VectorMultiplier32Unsigned.sv
LaneMul.sv
QDS.sv
CarrySaveAdder_28.sv
SqrtIter.sv
CarrySaveAdder_10.sv
QDS_1.sv
CarrySaveAdder_38.sv
OTF.sv
SRT16Iter.sv
Abs.sv
RoundingUnit.sv
SRTFPWrapper.sv
LaneDivFP.sv
LaneFFO.sv
LanePopCount.sv
OtherUnit.sv
MulAddRecFNToRaw_preMul_e8_s24.sv
MulAddRecFNToRaw_postMul_e8_s24.sv
MulAddRecFN_e8_s24.sv
RoundAnyRawFNToRecFN_ie6_is32_oe8_os24.sv
INToRecFN_i32_e8_s24.sv
RecFNToIN_e8_s24_i32.sv
Rec7LUT.sv
Rec7Fn.sv
Rsqrt7LUT.sv
Rsqrt7Fn.sv
Verification/Assert/LaneFloat_Verification_Assert.sv
LaneFloat.sv
Arbiter4_SlotRequestToVFU.sv
Arbiter1_SlotRequestToVFU.sv
ram_4x3.sv
Queue4_UInt3.sv
Distributor.sv
Queue1_VRFWriteRequest_2.sv
Arbiter1_VRFReadRequest.sv
Verification/Assert/Lane_Verification_Assert.sv
Lane.sv
T1.sv
AXI4SlaveAgent.sv
AXI4SlaveAgent_1.sv
Verification/Assert/TestBench_Verification_Assert.sv
Verification/TestBench_Verification.sv
TestBench.sv
```
</details>

<details>

<summary>Content of the directory</summary>

```
├── Abs32.sv
├── Abs.sv
├── AddRawFN.sv
├── AddRecFN.sv
├── Arbiter1_SlotRequestToVFU.sv
├── Arbiter1_VRFReadRequest.sv
├── Arbiter4_SlotRequestToVFU.sv
├── AXI4SlaveAgent_1.sv
├── AXI4SlaveAgent.sv
├── CarrySaveAdder_10.sv
├── CarrySaveAdder_28.sv
├── CarrySaveAdder_38.sv
├── ChainingCheck.sv
├── ClockGen.sv
├── CompareRecFN.sv
├── CrossReadUnit.sv
├── CSA42_1.sv
├── CSA42_2.sv
├── CSA42_3.sv
├── CSA42_4.sv
├── CSA42_5.sv
├── CSA42_6.sv
├── CSA42.sv
├── CSACompressor4_2.sv
├── Distributor.sv
├── extern_modules.sv
├── filelist.f
├── firrtl_black_box_resource_files.f
├── FloatAdder.sv
├── FloatCompare.sv
├── INToRecFN_i32_e8_s24.sv
├── LaneAdder.sv
├── LaneBitLogic.sv
├── LaneDivFP.sv
├── LaneExecutionBridge_1.sv
├── LaneExecutionBridge_2.sv
├── LaneExecutionBridge_3.sv
├── LaneExecutionBridge.sv
├── LaneFFO.sv
├── LaneFloat.sv
├── LaneLogic.sv
├── LaneMul.sv
├── LanePopCount.sv
├── LaneShifter.sv
├── LaneStage0_1.sv
├── LaneStage0.sv
├── LaneStage1_1.sv
├── LaneStage1.sv
├── LaneStage2_1.sv
├── LaneStage2.sv
├── LaneStage3_1.sv
├── LaneStage3.sv
├── Lane.sv
├── LoadUnit.sv
├── LSU.sv
├── MaskedLogic.sv
├── MaskedWrite.sv
├── MulAddRecFN_e8_s24.sv
├── MulAddRecFNToRaw_postMul_e8_s24.sv
├── MulAddRecFNToRaw_preMul_e8_s24.sv
├── Multiplier16.sv
├── OTF.sv
├── OtherUnit.sv
├── QDS_1.sv
├── QDS.sv
├── Queue14_LaneStage1Enqueue_1.sv
├── Queue14_LaneStage1Enqueue.sv
├── Queue14_UInt5.sv
├── Queue16_UInt8.sv
├── Queue17_UInt32.sv
├── Queue1_UInt64.sv
├── Queue1_VRFWriteRequest_2.sv
├── Queue1_VRFWriteRequest.sv
├── Queue2_MemWrite.sv
├── Queue2_SimpleMemWrite.sv
├── Queue2_UInt32_8.sv
├── Queue2_UInt32.sv
├── Queue4_LaneExecuteResponse_1.sv
├── Queue4_LaneExecuteResponse.sv
├── Queue4_LaneExecuteStage_1.sv
├── Queue4_LaneExecuteStage.sv
├── Queue4_ReadBusData.sv
├── Queue4_SimpleAccessStage1.sv
├── Queue4_UInt1.sv
├── Queue4_UInt32.sv
├── Queue4_UInt3.sv
├── Queue4_VRFReadQueueEntry.sv
├── Queue4_VRFWriteRequest_2.sv
├── Queue4_VRFWriteRequest.sv
├── Queue5_UInt3.sv
├── Queue6_ExecutionBridgeRecordQueue_1.sv
├── Queue6_ExecutionBridgeRecordQueue.sv
├── Queue96_LSUWriteQueueBundle.sv
├── ram_14x157.sv
├── ram_14x158.sv
├── ram_14x5.sv
├── ram_16x8.sv
├── ram_17x32.sv
├── ram_2x325.sv
├── ram_2x32.sv
├── ram_2x78.sv
├── ram_4x104.sv
├── ram_4x105.sv
├── ram_4x17.sv
├── ram_4x1.sv
├── ram_4x32.sv
├── ram_4x38.sv
├── ram_4x3.sv
├── ram_4x46.sv
├── ram_4x72.sv
├── ram_5x3.sv
├── ram_6x103.sv
├── ram_6x104.sv
├── ram_96x54.sv
├── ReadStageRRArbiter_1.sv
├── ReadStageRRArbiter.sv
├── Rec7Fn.sv
├── Rec7LUT.sv
├── RecFNToIN_e8_s24_i32.sv
├── ReduceAdder.sv
├── rfVec_rf_sram_64x36.sv
├── RoundAnyRawFNToRecFN_ie6_is32_oe8_os24.sv
├── RoundAnyRawFNToRecFN_ie8_is26_oe8_os24.sv
├── RoundingUnit.sv
├── RoundRawFNToRecFN_e8_s24.sv
├── Rsqrt7Fn.sv
├── Rsqrt7LUT.sv
├── SimpleAccessUnit.sv
├── SlotTokenManager.sv
├── SqrtIter.sv
├── SRT16Iter.sv
├── SRTFPWrapper.sv
├── StoreUnit.sv
├── T1.sv
├── TestBench.sv
├── VectorAdder32.sv
├── VectorAdder64.sv
├── VectorDecoder.sv
├── VectorMultiplier32Unsigned.sv
├── Verification
│   ├── Assert
│   │   ├── LaneExecutionBridge_1_Verification_Assert.sv
│   │   ├── LaneExecutionBridge_2_Verification_Assert.sv
│   │   ├── LaneExecutionBridge_3_Verification_Assert.sv
│   │   ├── LaneExecutionBridge_Verification_Assert.sv
│   │   ├── LaneFloat_Verification_Assert.sv
│   │   ├── LaneStage1_Verification_Assert.sv
│   │   ├── Lane_Verification_Assert.sv
│   │   ├── layers_TestBench_Verification_Assert.sv
│   │   ├── SimpleAccessUnit_Verification_Assert.sv
│   │   ├── StoreUnit_Verification_Assert.sv
│   │   ├── TestBench_Verification_Assert.sv
│   │   ├── VrfReadPipe_1_Verification_Assert.sv
│   │   ├── VrfReadPipe_Verification_Assert.sv
│   │   └── VRF_Verification_Assert.sv
│   ├── layers_TestBench_Verification.sv
│   └── TestBench_Verification.sv
├── VrfReadPipe_1.sv
├── VrfReadPipe.sv
├── VRF.sv
└── WriteCheck.sv
```

</details>

# Reproduce

Clone [dependencies-bumping-2024-08-21](https://github.com/chipsalliance/t1/tree/dependencies-bumping-2024-08-21)

Run `nix build -L .#t1.blastoise.ip.verilator-emu-rtl -o result-rtl`

The `result-rtl` directory will contains generated SystemVerilog code and filelist.f

sequencer, 2024-08-22:
cc @SpriteOvO

SpriteOvO, 2024-08-22:
> cc @SpriteOvO, I wonder if it's possible for layer to have their own `filelist.f`, e.g. `filelist.Verification.Assert.f`.

Is there a specific use case for this?

seldridge, 2024-08-22:
So there are a number of ways to handle this: (1) include the bind files in the compilation, (2) use `--enable-layers=Verification,Verification.Assert,Verification.Assume,Verification.Cover` to cause CIRCT to specialize these layers (which effectively undoes the changes in Chisel), or (3) make changes to the filelist / add more filelists.

I've handled this on the Chisel side for ChiselSim by adding the ability to customize what layers are enabled. The layers have predictable names and you can pattern match them based on what layers you want to enable.

We're currently using (2) internally while I work on flows to allow users to choose what layers they want/need for a specific test or use case.

seldridge, 2024-08-22:
Also, I should clarify that this change is _coming from Chisel_, specifically this: https://github.com/chipsalliance/chisel/commit/42f8a046ba199edc8ca651d044c1df170836d570

Nothing has changed with CIRCT/firtool regarding these statements.

unlsycn, 2024-08-25:
> > cc @SpriteOvO, I wonder if it's possible for layer to have their own `filelist.f`, e.g. `filelist.Verification.Assert.f`.抄送@SpriteOvO ，我想知道层是否可以拥有自己的`filelist.f` ，例如`filelist.Verification.Assert.f` 。
> 
> Is there a specific use case for this?这有具体的用例吗？

For example, verilator does not fully support assertion statements, and we can determine the layers to specific tools with a per-layer filelist.