llvm/circt #7586: [FIRRTL] Firtool may crash when compiling a certain fir file.

Siudya, 2024-09-05:
## Environment
OS: Ubuntu 22.04
firtool: 1.62.0 ~ 1.75.0
mill: 0.11.2
chisel: 6.5.0
firrtl: 3.3.0

## Repeat the crash
```bash
git clone https://github.com/Siudya/riscv-boom
cd riscv-boom
make chirrtl # success
make firrtl # success
make hw # failed
make verilog # failed
make systemverilog # failed
make rtl # failed
```

The failure log is as following:
```
ExitCode:
139
STDOUT:

STDERR:
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool   0x00000000006ffed7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 39
1  firtool   0x00000000006fdc8e llvm::sys::RunSignalHandlers() + 238
2  firtool   0x000000000070080f
3  libc.so.6 0x0000753698a42520
4  firtool   0x0000000000b253dd
5  firtool   0x0000000000b1d6af
6  firtool   0x0000000000b1837a
7  firtool   0x0000000000b3ce19
8  firtool   0x00000000007673dc
9  firtool   0x000000000113dcfc llvm::ThreadPool::processTasks(llvm::ThreadPoolTaskGroup*) + 828
10 firtool   0x000000000113e85c
11 libc.so.6 0x0000753698a94ac3
12 libc.so.6 0x0000753698b26850
```

## Other results
I have tried manually call firtool to compile imtermediate files and I got these:
```bash
firtool -o RawTileTop.fir.mlir --format fir --disable-annotation-unknown --ir-fir build/chirrtl/RawTileTop.fir # success
firtool -o RawTileTop.hw.mlir --format fir --disable-annotation-unknown --ir-hw build/chirrtl/RawTileTop.fir # failed
firtool -o RawTileTop.sv --format fir --disable-annotation-unknown --verilog build/chirrtl/RawTileTop.fir # failed

firtool -o RawTileTop.hw.mlir --format mlir --ir-hw build/firrtl/RawTileTop.fir.mlir # success
firtool -o RawTileTop.sv --format mlir --verilog build/firrtl/RawTileTop.fir.mlir # success
```
It seems firtool will crash somewhere when translate fir to hw-ir.

dtzSiFive, 2024-09-05:
This works with latest firtool (built from main anyway), modulo reverting #7030 for the printf-encoded asserts, so looks like a bug we've fixed since then.

Running with older firtool, looks like this fails in LowerToHW calling tryCopyName -- specifically what's fixed by #7358.

Yep, reverting the fix from 7358 causes latest firtool to crash as well.

Is building a custom firtool an option (either cherry-pick 7358 onto your release of choice, or use latest restoring 7030's deprecation of the printf-encoded verification statements), at least until upgrading Chisel version is possible/complete?

Siudya, 2024-09-06:
Recompiling a firtool is not an option for me. For now, I will use a workaround flow, that compiles Chisel to fir then uses firtool 1.62.0 to generate RTL. Is it reasonable that you release a firtool 1.62.1 merging the fix patch, as the latest release of Chisel still uses firtool 1.62.0?

closed 2024-09-09 by Siudya

dtzSiFive, 2024-09-09:
Cut a 1.62.1 with that fix: https://github.com/llvm/circt/releases/tag/firtool-1.62.1 .

jackkoenig, 2024-09-09:
@Siudya Chisel 6.x now uses firtool 1.62.1, this should be released as part of chisel 6.6.0 which I expect to release in the next week or two. I am waiting for the imminent release of Scala 2.13.15 and I would like Chisel 6.6.0 to support it.

You can try out Chisel with this firtool by using the current SNAPSHOT `6.5.0+6-120d36d8-SNAPSHOT`, or by manually adding `org.chipsalliance:llvm-firtool:1.62.1` as a dependency in your Scala build.