llvm/circt #7628: [MooreToCore] Support string constants

fabianschuiki, 2024-09-25:
Add a conversion for the `moore.string_constant` op to MooreToCore: https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+11.10+string_bit_array

jpinot, 2024-10-19:
Hey @fabianschuiki, I'd like to give it a shot! Could you assign me?

fabianschuiki, 2024-10-21:
Sure thing! Thanks a lot for taking a stab at this :smiley:. Let me know if you run into any trouble.

jpinot, 2024-10-27:
Hi @fabianschuiki! Sorry for the delayed response—it took me a bit to dive into the CIRCT fundamentals to get oriented :smiley_cat:. To tackle this, I referenced the [Moore test results](https://chipsalliance.github.io/sv-tests-results/?v=moore+11.10+string_bit_array) and tried [converting `moore::StringConstantOp` to `hw::ConstantOp` by translating `StringRef` into a bit vector](https://github.com/jpinot/circt/commit/a1747a3d59dba8124b4f820c358c87d3168511d9), also [I tryed using hw::ArrayCreateOp](https://github.com/jpinot/circt/commit/67c07c95a94cab240485d5ec0659aee71d3e79ca). Here’s the resulting LLHD:

```sh
$ ./bin/circt-verilog string_bit_array.sv
module {
  hw.module @top() {
    %0 = llhd.constant_time <0ns, 0d, 1e>
    %c0_i80 = hw.constant 0 : i80
    %c1415934836_i32 = hw.constant 1415934836 : i32
    %c0_i112 = hw.constant 0 : i112
    %a = llhd.sig %c0_i112 : i112
    llhd.process {
      %1 = comb.concat %c0_i80, %c1415934836_i32 : i80, i32
      llhd.drv %a, %1 after %0 : !hw.inout<i112>
      llhd.halt
    }
    hw.output
  }
}
```

It seems that this module should work as expected since it represents a constant string, which shouldn’t be modified. Still, I’d really appreciate your thoughts on (a) what the expected correct output should look like to confirm I’m on track, and (b) whether there might be a cleaner way to handle the `StringRef` to bit vector conversion. So far, I’ve only managed to iterate through the string and concatenate the bytes, but I’d love to know if there’s a more elegant approach.

Thanks so much for your help guidance!

fabianschuiki, 2024-10-27:
Hey @jpinot, this looks really nice! I think your approach of converting the characters in the string into a large `hw.constant` is great!

jpinot, 2024-11-06:
Hey @fabianschuiki, once again thanks for your support :partying_face: . Should this issue be close?

fabianschuiki, 2024-11-06:
Yeah sounds great! Thanks for landing that PR :smiley:

closed 2024-11-06 by fabianschuiki