llvm/circt #7630: [MooreToCore] Support assert, assume, cover ops

fabianschuiki, 2024-09-25:
Add conversion patterns from `moore.{assert,assume,cover}` to the corresponding `verif.*` op in MooreToConv: https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+16.2+assert_test

mingzheTerapines, 2024-10-14:
@fabianschuiki  
https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+16.2+assert_test
and
https://chipsalliance.github.io/sv-tests-results/?v=circt_verilog+16.4+assert0_test
all cleared, happy to see this issue could be closed =)

fabianschuiki, 2024-10-14:
Very exciting :partying_face:! Thanks for adding support for this in #7681 :smile_cat:

closed 2024-10-14 by fabianschuiki

fabianschuiki, 2024-10-14:
Also looks like we're up to 42% of tests passing now: https://chipsalliance.github.io/sv-tests-results/history :tada: