llvm/circt #7637: [Seq] Move SeqToSV tests from `test/Dialect/Seq` to `test/Conversion/SeqToSV`

uenoku, 2024-09-26:
Currently SeqToSv tests are scattered in `test/Dialect/Seq` but they should be put under `test/Conversion/SeqToSV`.

brown-eric, 2024-09-27:
I'd love to help with this. What tests do you want moved?

maerhart, 2024-10-02:
Thanks @brown-eric! The ones that invoke a pass called `--lower-seq-to-sv` should be moved.

Rahul-2006, 2025-01-16:
Hi @uenoku and contributors,

I would like to work on this issue as my first contribution to the CIRCT project or LLVM. I have already reviewed the codebase and the relevant directories, and I understand the steps required to move the `SeqToSV` tests from `test/Dialect/Seq` to `test/Conversion/SeqToSV`.

Could you please assign this issue to me? I will submit a pull request shortly after implementing and testing the changes.

Thank you,  
[Your Name]

Rahul-2006, 2025-01-16:
Hi @uenoku , @maerhart and contributors,

I would like to work on this issue as my first contribution to the CIRCT project or LLVM. I have already reviewed the codebase and the relevant directories, and I understand the steps required to move the SeqToSV tests from test/Dialect/Seq to test/Conversion/SeqToSV.

Could you please assign this issue to me? I will submit a pull request shortly after implementing and testing the changes.

Thank you,
Rahul

closed 2026-08-25 by uenoku