llvm/circt #7665: [LLHD][ProcessLowering] Incorrectly inlines aliasing drives

maerhart, 2024-10-04:
The ProcessLowering pass checks whether an `llhd.process` operation is a combinational process and inlines it into the parent module if that's the case.

However, it doesn't check for aliasing drives. There can be a series of blocking drives where the first one does a general default assignment to the entire signal, and subsequent conditional drives, aliasing part of the signal, override parts. Since there is no fixed execution order inside modules, these must be combined before inlining. ProcessLowering should check for that and not perform the inlining in such cases.

The TemporalCodeMotion pass combines drives to the same signal to some degree but doesn't consider subelement aliasing. One option to support this is to perform SROA before TemporalCodeMotion.

chenbo-again, 2025-02-25:
When I ran `circt-verilog --ir-llhd %s` to the follow code.
```verilog
module combo (  input   a,  output reg b);
  always @(*) begin
    b = 1'b0;
    b = 1'b1;
  end
endmodule
```
I get 
```mlir
module {
  hw.module @combo(in %a : i1, out b : i32) {
    %0 = llhd.constant_time <0ns, 0d, 1e>
    %c0_i32 = hw.constant 0 : i32
    %false = hw.constant false
    %a_0 = llhd.sig name "a" %false : i1
    %b = llhd.sig %c0_i32 : i32
    llhd.drv %a_0, %a after %0 : !hw.inout<i1>
    %1 = llhd.prb %b : !hw.inout<i32>
    hw.output %1 : i32
  }
}
```
It seems to be wrong, is that the same bug with this issue? I also tried `circt-opt --sroa %s`, and it did not help.

maerhart, 2025-02-25:
No, that's a different issue.

Strictly according to IEEE 1800-2017, the always procedure in your example is never executed because the implicit sensitivity list is specified to only include signals that are read inside the procedure. As there is no signal read in your example the procedure is sensitive to nothing and thus never triggered.

On current main this example produces the following when running the entire circt-verilog pipeline
```mlir
  hw.module @combo(in %a : i1, out b : i1) {
    %false = hw.constant false
    hw.output %false : i1
  }
```
So, it just returns the value initially assigned to signal 'b' (i.e., the initialization value of the variable, not the first assignment in your example), which is 0 by default (because we only support 2-state for now).

Most tools, however, treat the `always @(*)` procedures like `always_comb` and thus your procedure is executed exactly once by most tools. This means the output would be a constant 1 then. We are planning to add a CLI option to circt-verilog to enable this behavior.

chenbo-again, 2025-02-25:
Thank you for your reply, this testcase is originally from another case like below:
```
module combo (  input   a,  output reg b);
  always @(*) begin
    b = 1'b0;
    if(a) begin
      b = 1'b1;
    end
  end
endmodule
```
```
module {
  hw.module @combo(in %a : i1, out b : i1) {
    %0 = llhd.constant_time <0ns, 0d, 1e>
    %true = hw.constant true
    %false = hw.constant false
    %a_0 = llhd.sig name "a" %false : i1
    %1 = llhd.prb %a_0 : !hw.inout<i1>
    %b = llhd.sig %false : i1
    llhd.process {
      cf.br ^bb1
    ^bb1:  // 2 preds: ^bb0, ^bb2
      llhd.wait (%1 : i1), ^bb2
    ^bb2:  // pred: ^bb1
      %3 = llhd.prb %a_0 : !hw.inout<i1>
      %4 = comb.and %3, %true : i1
      %5 = comb.or %4, %false : i1
      %6 = comb.mux %5, %true, %false : i1
      %7 = comb.or %4, %false, %true : i1
      llhd.drv %b, %6 after %0 if %7 : !hw.inout<i1>
      cf.br ^bb1
    }
    llhd.drv %a_0, %a after %0 : !hw.inout<i1>
    %2 = llhd.prb %b : !hw.inout<i1>
    hw.output %2 : i1
  }
}
```
This case can not be properly desequentialized. It is very similar to the bug in this issue, but I'm not very sure this one can be solved by sroa pass.