llvm/circt #7682: [Calyx] IfOp's group has to be a CombGroupOp?

jiahanxie353, 2024-10-08:
Does `IfOp`'s `Group` has to be a `CombGroupOp`, as required [here](https://github.com/llvm/circt/blob/e76d2644a7782f87497668efbc57e76365b887c1/lib/Dialect/Calyx/CalyxOps.cpp#L2228)?

What if we have some examples like:
```mlir
  func.func @main(%arg0 : i32) {
    %two = arith.constant 2: i32
    %rem = arith.remui %arg0, %two : i32
    %cond0 = arith.cmpi eq, %arg0, %rem : i32

    scf.if %cond0 {
      some operations
    } else {
      some operations
    }
```
where `%cond0` involves comparing the `out` of `remui_pipe_op`, which is [not combinational](https://github.com/llvm/circt/blob/e76d2644a7782f87497668efbc57e76365b887c1/lib/Conversion/SCFToCalyx/SCFToCalyx.cpp#L651)?

What was the original design criteria? I believe @mortbopet is the original author. Also tagging  @rachitnigam @cgyurgyik

cgyurgyik, 2024-10-08:
This was probably a requirement that existed when this was initially implemented: https://docs.calyxir.org/lang/ref.html#if

jiahanxie353, 2024-10-08:
> This was probably a requirement that existed when this was initially implemented: https://docs.calyxir.org/lang/ref.html#if

Ahh indeed, thanks for pointing that out!

Is it true that `if` statements with `with` `group`s  is still an ongoing discussion: https://github.com/calyxir/calyx/issues/1699?

Semantically, I believe my example makes sense, where one of the `lhs`/`rhs` of the comparison is a result of some operation, which can be either a combinational circuit or sequential ones (like `multi_pipe_op`/`divi_pipe_op`/`remui_pipe_op`, since the upstream tools should be unaware if an operation is implemented combinationally or sequentially in hardware?).

As separate question, is there any workaround over this Well-formedness check, so that we can still have `if %cond with comb_group`, and that `comb_group` somehow compares `remui_pipe_op.out` with `%arg0` combinationally?

rachitnigam, 2024-10-08:
No, for non-combinational computations, you need to manually perform the computation in a group, store the value in a register, and then use the value of the register to perform the computation. For example:
```
compute_cond; // computes rem and store in reg
if reg.out { ... } else { ... }
```

cgyurgyik, 2025-03-18:
This is good to close @jiahanxie353 ?

jiahanxie353, 2025-03-18:
> This is good to close [@jiahanxie353](https://github.com/jiahanxie353) ?

Yes!

closed 2025-03-18 by jiahanxie353