llvm/circt #7698: [ExportVerilog][COMB] Crash in circt-opt: Variadic Operand Handling in ExportVerilog

YangWiz, 2024-10-13:
**Summary:** When running circt-opt, I encountered an assertion failure in ExportVerilog.cpp related to handling of a comb::AndOp operation. The tool crashes with the following error:
```
Assertion `op.getNumOperands() == 2 && "prelowering should handle variadics"' failed.
```
The documentation didn’t mention that I should pre-lower the variadics before using the [comb](https://circt.llvm.org/docs/Dialects/Comb/), so I think it's a bug. Additionally, the crash report suggests submitting a bug report:
"PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace."

The command I used is: 
```
circt-opt -convert-fsm-to-sv --lower-seq-to-sv -export-verilog
```

CIRCT version: commit 33c35fff11c79a3d3c0f78120c0296a700db2cfa
OS version: Ubuntu 22.04

The crash backtrace is:
```
#0 0x00005a6218015508 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/yanghoo/Code/circt/llvm/llvm/lib/Support/Unix/Signals.inc:723:22
  #1 0x00005a6218015929 PrintStackTraceSignalHandler(void*) /home/yanghoo/Code/circt/llvm/llvm/lib/Support/Unix/Signals.inc:798:1
  #2 0x00005a6218012d79 llvm::sys::RunSignalHandlers() /home/yanghoo/Code/circt/llvm/llvm/lib/Support/Signals.cpp:105:20
  #3 0x00005a6218014da0 SignalHandler(int) /home/yanghoo/Code/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
  #4 0x0000704278442520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
  #5 0x00007042784969fc pthread_kill (/lib/x86_64-linux-gnu/libc.so.6+0x969fc)
  #6 0x0000704278442476 gsignal (/lib/x86_64-linux-gnu/libc.so.6+0x42476)
  #7 0x00007042784287f3 abort (/lib/x86_64-linux-gnu/libc.so.6+0x287f3)
  #8 0x000070427842871b (/lib/x86_64-linux-gnu/libc.so.6+0x2871b)
  #9 0x0000704278439e96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
 #10 0x00005a621883e58b (anonymous namespace)::ExprEmitter::visitComb(circt::comb::AndOp) /home/yanghoo/Code/circt/lib/Conversion/ExportVerilog/ExportVerilog.cpp:2412:22
 #11 0x00005a6218889b0d (anonymous namespace)::SubExprInfo circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::comb::AndOp>(auto) const /home/yanghoo/Code/circt/include/circt/Dialect/Comb/CombVisitors.h:42:13
 #12 0x00005a62188896e7 llvm::TypeSwitch<mlir::Operation*, (anonymous namespace)::SubExprInfo>& llvm::TypeSwitch<mlir::Operation*, (anonymous namespace)::SubExprInfo>::Case<circt::comb::AndOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::'lambda'(auto)&) /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:102:28
 #13 0x00005a621886214e Case<circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:46:40
 #14 0x00005a621886214e Case<circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #15 0x00005a621886214e Case<circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #16 0x00005a621886214e Case<circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #17 0x00005a621886214e Case<circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #18 0x00005a621886214e Case<circt::comb::ModUOp, circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #19 0x00005a621886214e Case<circt::comb::DivSOp, circt::comb::ModUOp, circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #20 0x00005a621886214e Case<circt::comb::DivUOp, circt::comb::DivSOp, circt::comb::ModUOp, circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #21 0x00005a621886214e Case<circt::comb::MulOp, circt::comb::DivUOp, circt::comb::DivSOp, circt::comb::ModUOp, circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
 #22 0x00005a621886214e Case<circt::comb::SubOp, circt::comb::MulOp, circt::comb::DivUOp, circt::comb::DivSOp, circt::comb::ModUOp, circt::comb::ModSOp, circt::comb::ShlOp, circt::comb::ShrUOp, circt::comb::ShrSOp, circt::comb::AndOp, circt::comb::OrOp, circt::comb::XorOp, circt::comb::ICmpOp, circt::comb::ParityOp, circt::comb::ConcatOp, circt::comb::ReplicateOp, circt::comb::ExtractOp, circt::comb::MuxOp, circt::comb::CombinationalVisitor<(anonymous namespace)::ExprEmitter, (anonymous namespace)::SubExprInfo>::dispatchCombinationalVisitor(mlir::Operation*)::<lambda(auto:21)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/Ty [... cut ...] sv::CoverPropertyOp, circt::sv::BindOp, circt::sv::StopOp, circt::sv::FinishOp, circt::sv::ExitOp, circt::sv::FatalOp, circt::sv::ErrorOp, circt::sv::WarningOp, circt::sv::InfoOp, circt::sv::ReadMemOp, circt::sv::GenerateOp, circt::sv::GenerateCaseOp, circt::sv::ForOp, circt::sv::SampledOp, circt::sv::Visitor<(anonymous namespace)::StmtEmitter, llvm::LogicalResult>::dispatchSVVisitor(mlir::Operation*)::<lambda(auto:48)>&> /home/yanghoo/Code/circt/llvm/llvm/include/llvm/ADT/TypeSwitch.h:47:42
```

uenoku, 2024-10-14:
Do you mind if you could share the input MLIR file? AndOp should have been splitted into variadic ops.

YangWiz, 2024-10-16:
@uenoku Thank you for your reply! I found that the AndOp has been split correctly, except when there is only one input, which makes the operation essentially meaningless. I have many comb.and operations in my code, and the crash report suggested that I should pre-lower the variadic operations. This leads me to believe that the comb.and variadic function doesn't work properly without accounting for the single-input condition in my code.

The code caused crash:
```
%fsm_output = comb.and %0 : i1
```

closed 2024-11-20 by YangWiz

7FM, 2025-08-26:
I am wondering why this issue was closed; the issue still persists.
Here is a minimal example:
```mlir
hw.module @PrelowerVariadics(in %a : i1, out b: i1) {
  %0 = comb.and %a : i1
  hw.output %0 : i1
}
```
Run with `circt-opt test.mlir -hw-legalize-modules -export-verilog`