llvm/circt #7722: [FIRRTL] Canonicalizer erases empty bundle connection

uenoku, 2024-10-22:
```
module {
  firrtl.circuit "top" {
    firrtl.module @top(in %in: !firrtl.bundle<foo: bundle<>, t: uint<8>>, out %out: !firrtl.bundle<foo: bundle<>, t: uint<8>>) {
      %0 = firrtl.subfield %in[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %1 = firrtl.subfield %out[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %2 = firrtl.subfield %in[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %3 = firrtl.subfield %out[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      firrtl.matchingconnect %3, %2 : !firrtl.uint<8>
    }
  }
}

$ circt-opt test.mlir -canonicalize
module {
  firrtl.circuit "top" {
    firrtl.module @top(in %in: !firrtl.bundle<foo: bundle<>, t: uint<8>>, out %out: !firrtl.bundle<foo: bundle<>, t: uint<8>>) {
      %0 = firrtl.subfield %in[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %1 = firrtl.subfield %out[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      firrtl.matchingconnect %1, %0 : !firrtl.uint<8>
    }
  }
}
```

dtzSiFive, 2024-10-22:
Altering the original/input to have a connection using the empty bundle:

```mlir
module {
  firrtl.circuit "top" {
    firrtl.module @top(in %in: !firrtl.bundle<foo: bundle<>, t: uint<8>>, out %out: !firrtl.bundle<foo: bundle<>, t: uint<8>>) {
      %0 = firrtl.subfield %in[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %1 = firrtl.subfield %out[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %2 = firrtl.subfield %in[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      %3 = firrtl.subfield %out[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
      firrtl.matchingconnect %3, %2 : !firrtl.uint<8>
      firrtl.matchingconnect %1, %0 : !firrtl.bundle<>
    }
  }
}
```

And the canonicalizer seems to preserve the connection?

uenoku, 2024-10-23:
My bad, it was actually LowerTypes and it is partially caused by my local modification to LowerTypes so I'll just close it. Thank you for checking this!
```
// -----// IR Dump After FlattenMemory (firrtl-flatten-memory) //----- //
firrtl.module @top(in %in: !firrtl.bundle<foo: bundle<>, t: uint<8>>, out %out: !firrtl.bundle<foo: bundle<>, t: uint<8>>) {
  %0 = firrtl.subfield %in[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
  %1 = firrtl.subfield %out[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
  %2 = firrtl.subfield %in[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
  %3 = firrtl.subfield %out[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
  firrtl.matchingconnect %3, %2 : !firrtl.uint<8>
  firrtl.matchingconnect %1, %0 : !firrtl.bundle<>
}

// -----// IR Dump After LowerFIRRTLTypes (firrtl-lower-types) //----- //
firrtl.circuit "top" {
  firrtl.module @top(in %in: !firrtl.bundle<foo: bundle<>, t: uint<8>>, out %out: !firrtl.bundle<foo: bundle<>, t: uint<8>>) {
    %0 = firrtl.subfield %in[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
    %1 = firrtl.subfield %out[foo] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
    %2 = firrtl.subfield %in[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
    %3 = firrtl.subfield %out[t] : !firrtl.bundle<foo: bundle<>, t: uint<8>>
    firrtl.matchingconnect %3, %2 : !firrtl.uint<8>
  }
}
```

closed 2024-10-23 by uenoku