llvm/circt #7735: [OM] Missing Verifier on Class Op return Type

seldridge, 2024-10-26:
Currently, there is no verifier that the type of an `om.class` correctly matches the new `om.class.fields` terminator.

Consider the following:

``` mlir
om.class @A(%arg: i1) {
  om.class.fields %arg : i1
}
```

Currently, `circt-opt` will accept this. However, the type should be verified to be:

``` mlir
om.class @A(%arg: i1) -> (a: i1) {
  om.class.fields %arg : i1
}
```

mikeurbach, 2024-10-28:
Thanks for filing this. There is also https://github.com/llvm/circt/issues/7078, but that is more about adding back a local verifier on `object.field`. I think this one is more about a local verifier on `class.fields`, which we should also have. @leonardt can you please take a look?

seldridge, 2024-10-29:
Fixed via: https://github.com/llvm/circt/commit/e08011253e00aa4ceda22788fc7617967f2bfb05

closed 2024-10-29 by seldridge