llvm/circt #7753: [circt-verilog] zero parameter at module declaration fails when used

dacsson, 2024-10-31:
I have a mmu.sv module for RISC-V implementation: 
```Verilog
module mmu #(
    parameter integer DEVICE_COUNT = 8
) (
...
output logic [31:2] dev_addr[DEVICE_COUNT]
...
```

that is used in top module like:

```Verilog
mmu #(
        .DEVICE_COUNT(8)
    ) mmu (
...
        .dev_addr(
        '{
            rom_addr,
            ram_addr[0],
            ram_addr[1],
            ram_addr[2],
            ram_addr[3],
            led_addr,
            mtime_addr,
            mtimecmp_addr
        }

...
```

when trying to compile with `circt-verilog mmu.sv` i get this error:

```bash
core/mmu.sv:17:34: error: value must be positive
    output logic [31:2] dev_addr[DEVICE_COUNT],
```

is it impossible to compile such module? why so?

hailongSun2000, 2024-11-01:
Hey, @dacsson :smiley:! Based on the SV IEEE Std 1800-2017, we must pass a single positive number to its dimension when we declare an unpacked array. Please review the following image:
[image]

hailongSun2000, 2024-11-01:
By the way, slang doesn't handle `logic [31:2] dev_addr[0]`, but handles `logic[31:2] dev_addr[0:-1]`;

closed 2026-10-08 by dacsson