llvm/circt #7826: [HandshakeToDC] Missing arith conversions and memory stuff

teqdruid, 2024-11-15:
HandshakeToHW also lowers the arith and malloc dialects. This is missing from the DC flow. It should probably get pulled out into a separate pass.

teqdruid, 2024-11-20:
For `arith`, I just found the `map-arith-to-comb`.

mortbopet, 2024-11-25:
> For arith, I just found the map-arith-to-comb pass.

Usage of this should be orthogonal to `HandshakeToDC`. I think the best thing to do here is to have a default fallback lowering of non-"special" operations in `HandshakeToDC` which inserts a join for it's operand (i.e. a [unit rate actor](https://github.com/llvm/circt/blob/87d10b7582247e3e0e78052e080e17871d81c2ce/lib/Conversion/HandshakeToHW/HandshakeToHW.cpp#L1114). `HandshakeToHW` does this [extremely poorly](https://github.com/llvm/circt/blob/87d10b7582247e3e0e78052e080e17871d81c2ce/lib/Conversion/HandshakeToHW/HandshakeToHW.cpp#L1922-L1935) since it was written before `map-arith-to-comb`... hence it essentially custom-implements `map-arith-to-comb`.

Unsure what you mean with `malloc` dialect; are you referring to the [handshake memory operations?](https://circt.llvm.org/docs/Dialects/Handshake/#handshakememory-circthandshakememoryop)

teqdruid, 2024-11-26:
> You are receiving this email because you authored the thread.

Sorry... by "malloc", I meant "memref".