llvm/circt #7835: [FIRRTL] ModulePrefixAnnotation not working properly on Vec size 1 typed smems

jackkoenig, 2024-11-19:
Similar issue to https://github.com/llvm/circt/issues/7788.

Consider the following FIRRTL:
```firrtl
FIRRTL version 4.0.0
circuit Top : %[[
  {
    "class":"chisel3.ModulePrefixAnnotation",
    "target": "~Top|Top>ram",
    "prefix":"Foo_"
  }
]]
  public module Top :
    input clock : Clock
    input raddr : UInt<6>
    input waddr : UInt<6>
    input wdata : UInt<8>[1]
    output rdata : UInt<8>[1]

    smem ram : UInt<8>[1][64]

    write mport w = ram[waddr], clock
    connect w, wdata

    node readAddr = raddr ; Avoid https://github.com/llvm/circt/issues/7834
    read mport r = ram[readAddr], clock
    connect rdata, r
```
If you run this with Firtool 1.93.1, you will get:
```verilog
module ram_0_64x8(
...
endmodule
module Top(
...
  ram_0_64x8 ram_0_ext (
  ...
endmodule
```

If you remove all of the `[1]` (to make the data type of the smem no longer a Vec of size 1), the resulting memory will be prefixed as expected:
```verilog
module Foo_ram_64x8(
...
endmodule
module Top(
...
  Foo_ram_64x8 ram_ext (
  ...
endmodule
```

**Note** this only applies to Vecs of size 1, things work properly if the Vec is size 2 or larger.

seldridge, 2024-11-19:
It looks like Lower Types is dropping the prefix.

seldridge, 2024-11-19:
Fixed with 8a3dd14efab31937dd0f17e907c0b5d59ef0cd11. With this applied, the Verilog is:

``` verilog
// Generated by CIRCT firtool-1.93.1-8-g6754869b9
// VCS coverage exclude_file
module Foo_ram_0_64x8(
  input  [5:0] R0_addr,
  input        R0_en,
               R0_clk,
  output [7:0] R0_data,
  input  [5:0] W0_addr,
  input        W0_en,
               W0_clk,
  input  [7:0] W0_data
);

  reg [7:0] Memory[0:63];
  reg       _R0_en_d0;
  reg [5:0] _R0_addr_d0;
  always @(posedge R0_clk) begin
    _R0_en_d0 <= R0_en;
    _R0_addr_d0 <= R0_addr;
  end // always @(posedge)
  always @(posedge W0_clk) begin
    if (W0_en & 1'h1)
      Memory[W0_addr] <= W0_data;
  end // always @(posedge)
  assign R0_data = _R0_en_d0 ? Memory[_R0_addr_d0] : 8'bx;
endmodule

module Top(
  input        clock,
  input  [5:0] raddr,
               waddr,
  input  [7:0] wdata_0,
  output [7:0] rdata_0
);

  Foo_ram_0_64x8 ram_0_ext (
    .R0_addr (raddr),
    .R0_en   (1'h1),
    .R0_clk  (clock),
    .R0_data (rdata_0),
    .W0_addr (waddr),
    .W0_en   (1'h1),
    .W0_clk  (clock),
    .W0_data (wdata_0)
  );
endmodule
```

closed 2024-11-19 by seldridge

jackkoenig, 2024-11-19:
Looks great, thank you!