llvm/circt #7863: [HWToBTOR2] Builds are Flaky

seldridge, 2024-11-21:
All builds seem to have destabilized after one of the recent changes to HWToBTOR2. The CI windows builds seem to hit this most regularly, though I've experienced it on Linux and macOS.

An example of the failure is in this build log: https://github.com/llvm/circt/actions/runs/11945441683/job/33298106177

Here is a macOS failure:

```
******************** TEST 'CIRCT :: Conversion/HWToBTOR2/errors.mlir' FAILED ********************
Exit Code: 134

Command Output (stdout):
--
1 sort bitvec 32
2 constd 1 
--
Command Output (stderr):
--
RUN: at line 1: /Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt /Users/schuylereldridge/repos/github.com/llvm/circt/test/Conversion/HWToBTOR2/errors.mlir --convert-hw-to-btor2 --verify-diagnostics --split-input-file -o /dev/null
+ /Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt /Users/schuylereldridge/repos/github.com/llvm/circt/test/Conversion/HWToBTOR2/errors.mlir --convert-hw-to-btor2 --verify-diagnostics --split-input-file -o /dev/null
Assertion failed: (GetNumBytesInBuffer() == 0 && "Current buffer is non-empty!"), function SetBufferAndMode, file raw_ostream.cpp, line 115.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
 #0 0x0000000104cce984 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000da984)
 #1 0x0000000104cccea0 llvm::sys::RunSignalHandlers() (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000d8ea0)
 #2 0x0000000104ccefd8 SignalHandler(int) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000dafd8)
 #3 0x0000000199356584 (/usr/lib/system/libsystem_platform.dylib+0x18047a584)
 #4 0x0000000199325c20 (/usr/lib/system/libsystem_pthread.dylib+0x180449c20)
 #5 0x0000000199232a30 (/usr/lib/system/libsystem_c.dylib+0x180356a30)
 #6 0x0000000199231d20 (/usr/lib/system/libsystem_c.dylib+0x180355d20)
 #7 0x000000010688f680 llvm::raw_ostream::SetBufferSize(unsigned long) (.cold.4) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x101c9b680)
 #8 0x0000000104cb3eec llvm::raw_ostream::SetBufferSize(unsigned long) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000bfeec)
 #9 0x0000000104cb47b8 llvm::raw_ostream::write(char const*, unsigned long) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000c07b8)
#10 0x0000000104c63428 void write_unsigned_impl<unsigned int>(llvm::raw_ostream&, unsigned int, unsigned long, llvm::IntegerStyle, bool) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x10006f428)
#11 0x0000000104cb402c llvm::raw_ostream::operator<<(unsigned long) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x1000c002c)
#12 0x0000000105b14888 (anonymous namespace)::ConvertHWToBTOR2Pass::genSort(llvm::StringRef, unsigned long) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x100f20888)
#13 0x0000000105b1378c void llvm::function_ref<void (mlir::Operation*)>::callback_fn<std::__1::enable_if<!llvm::is_one_of<circt::hw::HWModuleOp, mlir::Operation*, mlir::Region*, mlir::Block*>::value && std::is_same<void, void>::value, void>::type mlir::detail::walk<(mlir::WalkOrder)1, mlir::ForwardIterator, (anonymous namespace)::ConvertHWToBTOR2Pass::runOnOperation()::$_0, circt::hw::HWModuleOp, void>(mlir::Operation*, (anonymous namespace)::ConvertHWToBTOR2Pass::runOnOperation()::$_0&&)::'lambda'(mlir::Operation*)>(long, mlir::Operation*) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x100f1f78c)
#14 0x0000000105b130e0 (anonymous namespace)::ConvertHWToBTOR2Pass::runOnOperation() (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x100f1f0e0)
#15 0x00000001066a9770 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x101ab5770)
#16 0x00000001066a9f88 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x101ab5f88)
#17 0x00000001066ae03c auto void mlir::parallelForEach<std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0>(mlir::MLIRContext*, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0&&)::'lambda'(std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>&&)::operator()<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&>(std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>&&) const (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x101aba03c)
#18 0x00000001066ae16c std::__1::__function::__func<llvm::LogicalResult mlir::failableParallelForEach<std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, void mlir::parallelForEach<std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0>(mlir::MLIRContext*, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0&&)::'lambda'(std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>&&)>(mlir::MLIRContext*, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0&&)::'lambda'(), std::__1::allocator<llvm::LogicalResult mlir::failableParallelForEach<std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, void mlir::parallelForEach<std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0>(mlir::MLIRContext*, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0&&)::'lambda'(std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>&&)>(mlir::MLIRContext*, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, std::__1::__wrap_iter<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_0&&)::'lambda'()>, void ()>::operator()() (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x101aba16c)
#19 0x0000000105277124 std::__1::__deferred_assoc_state<void, std::__1::__async_func<std::__1::function<void ()>>>::__execute() (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x100683124)
#20 0x000000019924f548 (/usr/lib/libc++.1.dylib+0x180373548)
#21 0x0000000104c82808 llvm::StdThreadPool::processTasks(llvm::ThreadPoolTaskGroup*) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x10008e808)
#22 0x0000000104c83fbc void* llvm::thread::ThreadProxy<std::__1::tuple<llvm::StdThreadPool::grow(int)::$_0>>(void*) (/Users/schuylereldridge/repos/github.com/llvm/circt/build/bin/circt-opt+0x10008ffbc)
#23 0x0000000199325f94 (/usr/lib/system/libsystem_pthread.dylib+0x180449f94)
#24 0x0000000199320d34 (/usr/lib/system/libsystem_pthread.dylib+0x180444d34)
[... cut ...]
# | #27 0x00007ff7be5911d3 (d:\a\circt\circt\build\bin\circt-opt.exe+0x6511d3)
# | #28 0x00007ff7bf9ee02a (d:\a\circt\circt\build\bin\circt-opt.exe+0x1aae02a)
# | #29 0x00007ff7bf9ed10f (d:\a\circt\circt\build\bin\circt-opt.exe+0x1aad10f)
# | #30 0x00007ff817b56b4c (C:\Windows\System32\ucrtbase.dll+0x26b4c)
# | #31 0x00007ff819fd4cb0 (C:\Windows\System32\KERNEL32.DLL+0x14cb0)
# | #32 0x00007ff81a2fedcb (C:\Windows\SYSTEM32\ntdll.dll+0x7edcb)
# `-----------------------------
# error: command failed with exit status: 0x80000003
```

TaoBi22, 2024-11-21:
Sorry about this! Hopefully @fzi-hielscher fixed this in #7864 - if not then I can always remove the errors test for now as that seems to be the addition that's causing all these problems. Was discussing with @maerhart and @fabianschuiki yesterday that the implementation probably wants to be more like the other export passes anyway, which hopefully would fix these weird issues.

seldridge, 2024-11-21:
Awesome, thanks! Yeah, that looks like the right fix.

closed 2024-11-21 by seldridge